Xinyuan-LilyGO/LilyGo-T-RGB #2: Cam

PlastiBots, 2022-10-12:


PlastiBots, 2022-10-26:
@LilyGO this was posted 14 days ago?  Any feedback on why this is failing?

mmMicky, 2022-11-07:
Sorry I forgot to mention that lv_conf.h needs to be placed in the same directory as LVGL. Otherwise, the path cannot be recognized. Please move this to the arduino/libs directory as well

DeepCoreB4, 2022-11-12:
Today I received the display. Very nice, i love it! Unfortunately, the flashing of the factory.ino does not work. respectively the upload works, but the sdcard is not recognized, the screen has a green background and the writing is unreadable.
Is it possible to use this board as Mp3 Bluetooth player , if so how ?
I would like the manufacturer to make a demo example, for example how to use the music example from LVGL with this board.

The only thing that works for me is the display.ino file...

Thanks for any help.
Kind regards DCb4

hpsaturn, 2022-11-13:
I fixed some issues (#4) with PlatformIO, maybe this PR  #5 that I did, fix the issues on Arduino IDE.

DeepCoreB4, 2022-11-15:
> Today I received the display. Very nice, i love it!
....
> .....
> I would like the manufacturer to make a demo example how to use the music example from LVGL with this board. Thanks for any help. Kind regards DCb4

@mmMicky thanks for implementing my wish.... :)

hpsaturn, 2022-11-18:
Thanks for merge my PR, is working in PlatformIO. I think that in Arduino IDE it should be works. For Arduino IDE users, maybe is a good moment to change to Pio, is more easy than Arduino IDE.

You only need:

```bash
cd example/factory
pio run --target upload
``` 
and PlatformIO does all for you :D

PD: Plus, if you want, also you haven't to install VSCode, but is recommended.

PlastiBots, 2022-12-14:
Re-visiting my OP request... Managed to get LVGL to compile, but I cannot get many of the examples to compile.  Each fails in a different way.  Looking at the **lv_benchmark** example.  Fails at line 256 saying it can't find lv_demo_bechmark() function, yet it exists in the  lv_demo_benchmark.c and lv_demo_benchmark.h files in the benchmark folder inside the example.       Unclear why this is failing?

`c:/users/dave/appdata/local/arduino15/packages/m5stack/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\lv_benchmark.ino.cpp.o:(.literal._Z5setupv+0x70): undefined reference to `lv_demo_benchmark'
c:/users/dave/appdata/local/arduino15/packages/m5stack/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\lv_benchmark.ino.cpp.o: in function `setup()':
C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\lv_benchmark/lv_benchmark.ino:256: undefined reference to `lv_demo_benchmark'
collect2.exe: error: ld returned 1 exit status`

mmMicky, 2022-12-15:
As far as I know Arduino cannot find the LVGL/Demos directory. So I have ported lv_benchmark in sibling directory. It should be searchable after compilation. I built it successfully using PlatformIO.

PlastiBots, 2022-12-15:
Nope.. Doesn't work.  Also ensured this was done per LVGL readme:   _In `lv_conf.h` or equivalent places set `LV_USE_DEMO_BENCHMARK 1`_

hpsaturn, 2022-12-15:
I'm not Arduino IDE user, I leaved it many years ago for the same, it has a bad support for libraries dependency resolver, no CLI, no CI, etc, but maybe first try with the more simple sample, maybe with the example "display.ino" and check the next things:

- Check that Arduino IDE is using the Arduino Espressif framework v2.0.5
- Check that it is using the `lib` directory, that means:  `Arduino_GFX  lv_conf.h  lvgl  TouchLib` libraries from this repo
- Of course this repo updated: `git pull` in the root
- Selected the board `ESP32-S3-DevKitC-1`
- And the board with this parameters: `cpu = esp32s3` and `memory_type = qio_opi`

At the end say that maybe is a good moment to leave Arduino IDE and migrate to PlatformIO, maybe you will remember me when you will do that, because Pio is more easy than Arduino IDE.

PlastiBots, 2022-12-16:
Thanks for the tips, but will respectfully disagree with you.  I've used Arduino since the beginning, have done 100's of projects using many libraries and find it an excellent, well supported system.  Hence, why I am looking for this device to provide better support.  Not moving over to any other IDE etc anytime soon.    Also, I have already done all those things you mentioned, but again thanks for suggesting them.   @mmMicky I noticed that you included lvgl libraries etc within this T-RGB library.  Is the intent that these should be moved out and used within the Arduino lib folder? Currently I have the latest LVGL from git installed in the arduino libraries folder. Is the expectation that the one you provided should be used? Currently most of the build errors I am seeing related to missing functions etc as if they can't be found in the lvgl lib.

hpsaturn, 2022-12-16:
Well, only for clarify, Arduino IDE is for beginners, is a good tool but for instance it hasn't freeze libraries, auto resolution of libraries, continuous integration CLI, build with pipe lines, and a large list of missing features, and also the issue that you mentioned, it can't specifies libraries targets, or set files before each build, or retrieve libraries from git submodules, etc. 

I'm glad that you found the issue, like you mentioned, it was the second point that I wrote. Yes, always we should use the libraries of this repo, while LilyGO port the changes to official libraries.

hpsaturn, 2022-12-16:
PD: We would close this issue? isn't it?

PlastiBots, 2022-12-16:
No, this is not ready to be closed.  Yet another example - the GFX.ino example also fails on compile. Even after using the embedded Arduino_GFX library there are still issues (see below).  The overarching issue here is that, for most makers like myself, LilyGo should want to ensure examples are easy places to start.  We are not all advanced programmers using advanced tools - and nor do we want to be.   This is hobby for most of us and like myself, I am sure many others get by just fine using the Arduino IDE. 
  However, this set of examples are not well enough documented to help folks like me pick up and start building on top.  I would like to use Arduino GFX (not LVGL), yet now I have another mountain to climb to just figure out how to get a simple (provided) sample working.  I think others would be less patient and give up.  The examples should have better documentation/wiki against each so it's easy to get them working - and to be clear, for the Arduino environment - which does have a large following regardless of what your personal beliefs/preferences are.  

Below are the errors with GFX.ino (Note that the example is calling for Arduino_ST7701_RGBPanel which isn't even in the drivers folder provided in the library).  I found a set from my other builds, but it still doesn't work.)    So, the below errors shows as a result of the drivers I placed there - which are clearly not correct. Prior, the error is simply the driver is not defined.

`In file included from C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/Arduino_GFX_Library.h:72,
                 from C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\GFX\GFX.ino:7:
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_ST7701_RGBPanel.h:17:51: error: redefinition of 'const uint8_t st7701_type1_init_operations []'
 static const uint8_t st7701_type1_init_operations[] = {
                                                   ^
In file included from C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/Arduino_GFX_Library.h:67,
                 from C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\GFX\GFX.ino:7:
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_RGB_Display.h:215:22: note: 'const uint8_t st7701_type1_init_operations [273]' previously defined here
 static const uint8_t st7701_type1_init_operations[] = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/Arduino_GFX_Library.h:72,
                 from C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\GFX\GFX.ino:7:
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_ST7701_RGBPanel.h:128:51: error: redefinition of 'const uint8_t st7701_type2_init_operations []'
 static const uint8_t st7701_type2_init_operations[] = {
                                                   ^
In file included from C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/Arduino_GFX_Library.h:67,
                 from C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\GFX\GFX.ino:7:
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_RGB_Display.h:327:22: note: 'const uint8_t st7701_type2_init_operations [261]' previously defined here
 static const uint8_t st7701_type2_init_operations[] = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/Arduino_GFX_Library.h:72,
                 from C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\GFX\GFX.ino:7:
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_ST7701_RGBPanel.h:237:51: error: redefinition of 'const uint8_t st7701_type3_init_operations []'
 static const uint8_t st7701_type3_init_operations[] = {
                                                   ^
In file included from C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/Arduino_GFX_Library.h:67,
                 from C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\GFX\GFX.ino:7:
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_RGB_Display.h:437:22: note: 'const uint8_t st7701_type3_init_operations [288]' previously defined here
 static const uint8_t st7701_type3_init_operations[] = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\GFX\GFX.ino:13:123: error: no matching function for call to 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int)'
     EXAMPLE_PIN_NUM_DATA13, EXAMPLE_PIN_NUM_DATA14, EXAMPLE_PIN_NUM_DATA15, EXAMPLE_PIN_NUM_DATA16, EXAMPLE_PIN_NUM_DATA17);
                                                                                                                           ^
In file included from C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/Arduino_GFX_Library.h:14,
                 from C:\Users\Dave\Documents\Arduino\libraries\T-RGB-main\example\GFX\GFX.ino:7:
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/databus/Arduino_ESP32RGBPanel.h:59:3: note: candidate: 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint1 [... cut ...]
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_ST7701_RGBPanel.h:352:7: note:   candidate expects 1 argument, 15 provided
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_ST7701_RGBPanel.h:352:7: note: candidate: 'constexpr Arduino_ST7701_RGBPanel::Arduino_ST7701_RGBPanel(Arduino_ST7701_RGBPanel&&)'
C:\Users\Dave\Documents\Arduino\libraries\Arduino_GFX\src/display/Arduino_ST7701_RGBPanel.h:352:7: note:   candidate expects 1 argument, 15 provided
`

DeepCoreB4, 2022-12-16:
> Well, only for clarify, Arduino IDE is for beginners, is a good tool but for instance it hasn't freeze libraries, auto resolution of libraries, continuous integration CLI, build with pipe lines, and [...]

I have e project an .ino file but cant use with platformio ... only with arduino.

How to use it with platformion....
It tells me en error platformio.ini not found!

How to implement en project from arduino??

I like VsCode with  platformio, it's faster then arduino.

PlastiBots, 2022-12-16:
@DeepCoreB4, please create your issue in it's own thread.  My thread is specific about examples working in Arduino IDE, not Platform.io.

hpsaturn, 2022-12-16:
> I have e project an .ino file but cant use with platformio ... only with arduino.
> How to use it with platformion.... It tells me en error platformio.ini not found!
> How to implement en project from arduino??

Well, is possible, exists different ways, for example you are able to use the command line of PlatformIO, `pio` command, or using the VSCode Editor with PlatformIO pluging. 

For me is more easy via command line, for init the project. After that you able to open it using the GUI with VSCode. Of course, also you able to use only GUI without command line, but here I will try to explain the general steps using the CLI: is:

- validate first that you have pio installed in your shell or power shell: `pio -h`
- create a empty directory and enter there, e.g: `mkdir yourproject` 
- then select your board, for that first search it, for instance `pio boards esp32dev`
- select the board and init the project, e.g: `pio init --board=esp32dev`
- put your ino file into the directory with name `src`
- connect your board, and compile the project and upload it: `pio run --target upload`

Troubleshoting:

Missing libraries. No problem, search it, e.g `pio pkg search u8g2` and install it, e.g: `pio pkg install -l olikraus/U8g2`

I have some videos that explain it, for example this in youtube:

[image]