Xinyuan-LilyGO/LilyGo-T-RGB #3: can't create img.h or gImg .c / factory.ino not working

DeepCoreB4, 2022-11-13:
Im trying tto add a Gif animation...

I ussed LVGL IMG Converter..but does not working for me.

which Program do you use for adding in  .c or .h files??

> Edit: Unfortunately, the flashing of the factory.ino does not work. Error: ..Documents\Arduino\T-RGB\example\factory\factory.ino:11:10: fatal error: TouchLib.h: No such file or directory #include "TouchLib.h" ^~~~~~~~~~~~ compilation terminated.
> 
> The only thing that works for me is the display.ino/GFX.ino file...
> 
> I would like the manufacturer to make a demo example how to use the music example from LVGL with this board. Thanks for any help. Kind regards DCb4

nikthefix, 2022-11-13:
I'm doing gifs with this board and it's working ok. The LVGL image converter set to 'raw' is a must.  Have you copied the lilygo Lib folder to your Arduino library folder?

DeepCoreB4, 2022-11-15:
>Yes, thank you for your reply. Now I tried the method "CF_RAW" unfortunately does not work. The gif seems to be too big. What is the ideal frame rate? My gif had 35 frames. Does the file have to be written in .h or in .c?
✌️

nikthefix, 2022-11-17:
@DeepCoreB4  The output file should be .c
Many gifs are not optimised and can be very big for not much reward. I recommend using this online tool to see if you can optimise, remove frames or crop:

https://ezgif.com

Make sure that the 'use gif library' switch is set in lv_config.h

You should be able to animate a 300x300px gif without the sketch failing due to lack of memory but your mileage may vary. If all fails then try the gif example in the LVGL directory.

closed 2024-02-26 by github-actions[bot]