Xinyuan-LilyGO/LilyGo-T-RGB #6: [request] The board, plastic spacer and battery 3d models

hpsaturn, 2022-11-13:
## Overview

I'm working in a 3d printer box model to protect the board. Is possible that I'll have the 3d models referenced in this images?

[image]
[image]
 
I have a initial version, but this models could help me a lot:

https://user-images.githubusercontent.com/423856/201533918-d268d12f-5b67-4009-ac89-074e4cd3fddd.mp4

[image]

Thanks in advanced

mmMicky, 2022-12-02:
The models provided by the repository can be used arbitrarily.
https://github.com/Xinyuan-LilyGO/T-RGB/tree/main/Shell_file

hpsaturn, 2022-12-03:
thanks @mmMicky I didn't know, I'm going to try to convert it to Linux.

PlastiBots, 2022-12-16:
Can you provide STL versions of the model pls?

PlastiBots, 2022-12-27:
Hi @mmMicky, can you provide STL versions please?

mmMicky, 2022-12-28:
I am not a structural designer. I don't know how to convert it,

PlastiBots, 2022-12-28:
The CAD package should be able to export it as STL.

hpsaturn, 2023-01-08:
I released a version of my box, maybe if anyone try to test it, any feedback is welcome:
[image]

https://twitter.com/hpsaturn/status/1612129582194343937

sandmen, 2023-01-08:
Looks awesome...
Thingiverse project contains one file/model with all parts, as one part.
I can not imaging, that I can put the T-RGB device into this model.

hpsaturn, 2023-01-14:
please, read the description or "summary" in the Thingiverse post..

euphi, 2023-02-20:
> The models provided by the repository can be used arbitrarily. https://github.com/Xinyuan-LilyGO/T-RGB/tree/main/Shell_file

It seems that the step file is the T-Watch model, not the T-RGB.

Can you provide the correct STEP file?

sicca95, 2023-06-22:
Hi, everyone. Could you upload the correct STEP file from which has been taken the view of the display.dxf file? Thanks

closed 2024-02-11 by hpsaturn