Xinyuan-LilyGO/LilyGo-T-RGB #7: Touch release can cause erratic behavior with some LVGL widgets

nikthefix, 2022-11-13:
The touch driver will send x=0, y=0 when touch is released. Users should be aware that this pair of values could be sent to an LVGL control (arc, slider etc) after the touch interrupt pin goes high causing erratic behavior. If the control is animated then intermediate values will be registered. To hold the last touch location before release just trap and ignore p.x and p.y = 0. For the circular display this is a good workaround. Could this be implemented in the examples to help users new to the board?

LilyGO, 2022-11-15:
It is not normal for touch to send x=0, y=0. The correct coordinates should be picked up on a touch screen.

tijntje, 2022-11-16:
I can confirm the behaviour nikthefix describes; 
In the factory code the touch position is shown in 2 colors on the screen, blue with 0,0 when not touched and in green, with the coordinates as text.

The problem is that it seems that the driver sends 0,0 when you release touch (as indicated with the blue 0,0) this gives stranges effects in LVGL widgets.

As a poc I've changed lv_touchpad_read in factory.ino to make lvgl work correctly.

static void lv_touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data) {

  touch_point_t p = {0};
  if (touch_pin_get_int) {
#if defined(TOUCH_MODULE_FT3267)
    uint8_t touch_points_num;
    ft3267_read_pos(&touch_points_num, &p.x, &p.y);
    if (p.x != 0 && p.y != 0)
    {
    data->point.x = p.x;
    data->point.y = p.y;
    }
#elif defined(TOUCH_MODULE_CST820)
    touch.read();
    TP_Point t = touch.getPoint(0);
    data->point.x = t.x;
    data->point.y = t.y;
#endif
    data->state = LV_INDEV_STATE_PR;
    touch_pin_get_int = false;
  } else {
    data->state = LV_INDEV_STATE_REL;
  }
      if (p.x != 0 && p.y != 0)
    {
  lv_msg_send(MSG_TOUCH_UPDATE, &p);
    }
}

nikthefix, 2022-11-17:
@tijntje My workaround was the same as yours. I have noticed that many other touch controllers exhibit the same behaviour - presumably to allow for polled release detection in the event that a touch interrupt pin is either not used or not available.

DeepCoreB4, 2022-12-11:
> tijntje

Yes that's the workaround for my problem with the Touch issue...now ist smooth like butter!!)

DeepCoreB4, 2022-12-11:
Now, how can hear the sound, over Bluetooth headset or Bluetooth SoundBox??

mmMicky, 2022-12-15:
It is a demo display of lvgl. It is mainly used to demonstrate this GUI function, and does not realize the playback of underlying music. If you need this function, you need to add the application code yourself.

euphi, 2022-12-29:
see #15  for fix .

closed 2024-02-12 by github-actions[bot]