Xinyuan-LilyGO/LilyGo-T-RGB #8: Unable to program

user0693, 2022-11-30:
Hey,

I've been following all the steps and reading all the documents and instructions, but nothing works. Every time I try and run it I get errors in Arduino IDE. Now my ESP32-S3 won't even turn on.

mmMicky, 2022-12-02:
Did it fail to compile? Or is there a problem when uploading?

gustavoportilla85, 2022-12-31:
I have the same problems in uploading with Arduino IDE, it does not compile any program

ihayri, 2022-12-31:
Try the "display" example that should get you started, that is my base code I am using and have added jpeg image viewer, touch gestures to put it to deep sleep and touch interrupt pin to wake it up again, since the board has no power switch. Also make sure you use the you only use the Arduino_GFX, lvgl, TouchLib libraries that it comes with, other versions do not work, and copy the lv_config.h file in the lvgl library folder. Also do not updae the libraries or it will break it again.

PlastiBots, 2022-12-31:
You can also try the [touch swipe](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/example/TouchGesturesTestingV2/TouchGesturesTestingV2.ino) sample I provided in the library here.  It doesn't use LVGL and you can install the Arduino GFX library from GIT rather than the one modified from here.  It should get you up and running.

euphi, 2023-01-02:
I created a library for easier use here https://github.com/fablabnbg/TRGBArduinoSupport .

It supports platformio and is not yet integrated into Arduino IDE. However, if you want to give platformio a try (it integrates well into Visual Studio Code or Eclipse CDT), this can be a nice library :-)

gustavoportilla85, 2023-01-02:
> 

Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'git@github.com:fablabnbg/TRGBArduinoSupport.git'

willba4, 2023-01-04:
i'm also unable to compile any of the examples.  i've installed the extra couple of libraries that are included in the official github page.

mmMicky, 2023-01-05:
> i'm also unable to compile any of the examples. i've installed the extra couple of libraries that are included in the official github page.

You need to try to compile the simplest blink example first to prove that your environment is fully built.

Revoxdyna, 2023-01-18:
I am not able to compile the 

> Try the "display" example that should get you started, that is my base code I am using and have added jpeg image viewer, touch gestures to put it to deep sleep and touch interrupt pin to wake it up [...]

Could you share the "lvgl" libraries here?
I am using arduino IDE 2.0.3. It does not come with the libraries of Arduino_GFX, lvgl, TouchLib .
After download those libraries and followed the setup instructions for lv_conf.h, I still can not compile the example of "factory". It showed the following error msg and obviously the `factory_ui.cpp` have some problems, especially the function of `lv_msg_subsribe_obj` (` error: 'lv_msg_subsribe_obj' was not declared in this scope`)

```
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp: In function 'void ui_begin()':
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:47:3: error: 'lv_msg_subsribe_obj' was not declared in this scope
   lv_msg_subsribe_obj(MSG_BAT_VOLT_UPDATE, bat_label, NULL);
   ^~~~~~~~~~~~~~~~~~~
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:47:3: note: suggested alternative: 'lv_img_set_size_mode'
   lv_msg_subsribe_obj(MSG_BAT_VOLT_UPDATE, bat_label, NULL);
   ^~~~~~~~~~~~~~~~~~~
   lv_img_set_size_mode
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp: In lambda function:
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:52:9: error: 'lv_msg_t' was not declared in this scope
         lv_msg_t *m = lv_event_get_msg(e);
         ^~~~~~~~
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:52:9: note: suggested alternative: 'lv_img_t'
         lv_msg_t *m = lv_event_get_msg(e);
         ^~~~~~~~
         lv_img_t
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:52:19: error: 'm' was not declared in this scope
         lv_msg_t *m = lv_event_get_msg(e);
                   ^
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:52:19: note: suggested alternative: 'tm'
         lv_msg_t *m = lv_event_get_msg(e);
                   ^
                   tm
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:52:23: error: 'lv_event_get_msg' was not declared in this scope
         lv_msg_t *m = lv_event_get_msg(e);
                       ^~~~~~~~~~~~~~~~
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:52:23: note: suggested alternative: 'lv_event_get_key'
         lv_msg_t *m = lv_event_get_msg(e);
                       ^~~~~~~~~~~~~~~~
                       lv_event_get_key
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:53:41: error: 'lv_msg_get_payload' was not declared in this scope
         const float *v = (const float *)lv_msg_get_payload(m);
                                         ^~~~~~~~~~~~~~~~~~
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:53:41: note: suggested alternative: 'lv_img_get_pivot'
         const float *v = (const float *)lv_msg_get_payload(m);
                                         ^~~~~~~~~~~~~~~~~~
                                         lv_img_get_pivot
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp: In function 'void ui_begin()':
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:56:7: error: 'LV_EVENT_MSG_RECEIVED' was not declared in this scope
       LV_EVENT_MSG_RECEIVED, NULL);
       ^~~~~~~~~~~~~~~~~~~~~
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:56:7: note: suggested alternative: 'LV_EVENT_RELEASED'
       LV_EVENT_MSG_RECEIVED, NULL);
       ^~~~~~~~~~~~~~~~~~~~~
       LV_EVENT_RELEASED
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp: In lambda function:
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:65:9: error: 'lv_msg_t' was not declared in this scope
         lv_msg_t *m = lv_event_get_msg(e);
         ^~~~~~~~
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:65:9: note: suggested alternative: 'lv_img_t'
         lv_msg_t *m = lv_event_get_msg(e);
         ^~~~~~~~
         lv_img_t
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:65:19: error: 'm' was not declared in this scope
         lv_msg_t *m = lv_event_get_msg(e);
                   ^
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:65:19: note: suggested alternative: 'tm'
         lv_msg_t *m = lv_event_get_msg(e);
                   ^
                   tm
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory_ui.cpp:65:23: error: 'lv_event_get_msg' was not declared in this scope
[... 68 lines cut ...]
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory.ino:188:3: error: 'touch' was not declared in this scope
   touch.init();
   ^~~~~
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory.ino: In function 'void loop()':
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory.ino:296:5: error: 'lv_msg_send' was not declared in this scope
     lv_msg_send(MSG_BAT_VOLT_UPDATE, &v);
     ^~~~~~~~~~~
[... cut ...]
   ^~~~~~~~~~~
C:\Users\USER\Desktop\LCD_2inch1_lillygo\T-RGB-main\example\factory\factory.ino:400:3: note: suggested alternative: 'lv_fs_seek'
   lv_msg_send(MSG_WIFI_UPDATE, str.c_str());
   ^~~~~~~~~~~
   lv_fs_seek
Multiple libraries were found for "Wire.h"
  Used: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\Wire
  Not used: C:\Users\USER\Documents\Arduino\libraries\Wire
exit status 1

Compilation error: 'lv_msg_subsribe_obj' was not declared in this scope```

furynick, 2023-09-28:
Same problem here, display example compiles and run without issue but factory example fails with "not declared" errors :
```
D:\Arduino\T-RGB-main\examples\factory\factory.ino: In function 'void lv_touchpad_read(lv_indev_drv_t*, lv_indev_data_t*)':
D:\Arduino\T-RGB-main\examples\factory\factory.ino:260:5: error: 'lv_msg_send' was not declared in this scope
     lv_msg_send(MSG_TOUCH_UPDATE, &p);
     ^~~~~~~~~~~
```

closed 2024-02-11 by github-actions[bot]

Robiv8, 2024-10-12:
Wy is this issue closed without update?
I have the same Problem! New LilyGO T3 S4 with Arduino IDE, all Librarys loaded and any of the examples are not compiled.
By the way: Blink OK ;-)