Xinyuan-LilyGO/LilyGo-T-RGB #9: [Newbie] - this device is Useless without real functions...

DeepCoreB4, 2022-11-30:
Hi..
How to connect this device with Bluetooth?
All these demos are useless for a newbie in this field.
The only thing that worked for me is playing gif animations.
Now I ask myself the question? is that all?
What should I do now? What do you use this device for?

Now I finally got the music demo version of LVGL to work.
It only worked with the VSCode Platformio, Arduino IDE didn't work for me.
**Touch does not work**. So I have no idea if playing the music is possible at all. 

How to learn that at all...without a functional working example??

mmMicky, 2022-12-01:
T-RGB uses ESP32S3 as the main control. This means that all the routines on github that support ESP32S3 do. It is impossible for me to collect all the examples on the internet. I can provide this board-specific driver such as the screen so that you can use it.

mmMicky, 2022-12-01:
As far as I know, there are two types of touch chips. If you're not sure, you can take turns uncommenting here.

https://github.com/Xinyuan-LilyGO/T-RGB/blob/0fb2bc26e2eeb1a661385384c3e59848f99068f8/example/factory/factory.ino#L16

DeepCoreB4, 2022-12-03:
Hello mmMicky, thank you very much for your answer.
.....>
mmMicky> It is impossible for me to collect all examples on the internet
Yes, I realize that you can't create a working demo script for every device you develop. But if you add a specific resource from another repo for the device, you should assume that it works!
> As far as I know there are two types of touch chips. If you're not sure, you can alternate here without commenting.
.....>

oK...I will try it
hmmm If I uncomment // #define TOUCH_MODULE_CST820 I get a compile error....

mmMicky, 2022-12-05:
Could you please add printf to print (X,Y) coordinates here to help me figure out the problem. 
https://github.com/Xinyuan-LilyGO/T-RGB/blob/0fb2bc26e2eeb1a661385384c3e59848f99068f8/example/lv_music/lv_music.ino#L139

DeepCoreB4, 2022-12-06:
> Could you please add printf to print (X,Y) coordinates here to help me figure out the problem.
> 
> https://github.com/Xinyuan-LilyGO/T-RGB/blob/0fb2bc26e2eeb1a661385384c3e59848f99068f8/example/lv_music/lv_music.ino#L139

OK, I added a printf to read out the coordinates....
with; 
    // to find a fix for touch issu............................
    Serial.printf("Touch_Points_num(x,y): %llucoord\n", &x, &y);
    if (x > 0 || y > 1) {
      Serial.print("x,pos--> ");
      Serial.print(x);
      Serial.print("\ny,pos--> ");
      Serial.println(y);
      Serial.println("...");
      Serial.printf("XL9535 0x%02x : 0x%02X \r\n", xl, data);
      Serial.println("End off touch_readPos < ...");
    }//.........................................................

In the Serial Monitor I get the following result:
>... Starting touch_readPos
Touch_Points_num(x,y): 4597663406542958208coord
x,pos--> 12080
y,pos--> 16334
...
XL9535 0x3fce2e84 : 0x3FCE2EE8
End off touch_readPos < ...

I don't know if this information is correct or helpful, or if the "printf" line is correct.

mmMicky, 2022-12-07:
> 

It should only be able to read the range from 0 to 480. Anything larger than this range is incorrect. There may be a communication error.

DeepCoreB4, 2022-12-08:
hmmm...ok

Now i have this Output;
 its change on every position i touch the screen....

```
**middle screen: center touch:**
Touch_Points_num(x,y): 4597663458082565772coord
x,pos--> 0
y,pos--> 271
...
End off touch_readPos ...< 

**relesed touch: middle center:**
>... Starting touch_readPos 
Touch_Points_num(x,y): 4597663458082565772coord
x,pos--> 128
y,pos--> 271

**right side midle screen:**
>... Starting touch_readPos 
Touch_Points_num(x,y): 4597663458082565772coord
x,pos--> 257
y,pos--> 284
...
End off touch_readPos ...<

**Left middle screen**
>... Starting touch_readPos 
Touch_Points_num(x,y): 4597663458082565772coord
x,pos--> 44
y,pos--> 286
...
End off touch_readPos ...<
```
what now?? what can i do with this data?

mmMicky, 2022-12-08:
This information appears to be correct. You can make GUI on screen. For example, the music example on the routine. The main function of touch is to interact with people.

DeepCoreB4, 2022-12-08:
Yes, this was a small task where I learned something. :)
But still not enough.

> "The main function of touch is to interact with people."
> Sorry, I don't quite understand what you mean!

**The question was:** 
What should I do with this information now.
How can I solve the touch problem now?

With this **_printf line_** I added, the board detects the touch point and position. 
The board seems to be a bit slower than before.

Now I can swipe left or right, but I had to keep my finger on the screen for a few seconds to change the desired action. 
The same goes for playing and pausing music.

So I added a number of milliseconds of delay(2000); to pause. This was too much and the board stopped responding at all! So I changed this to delay(200);

**Is this now the TouchFix??**
hmmm...works a bit better now! 
---------------------------------------------------

OK, next part; 
When I want to listen to the sound, i put on my sdcard,.. on my bluetooth speakers/headphones! How should I best proceed? do you have me advice ?

DeepCoreB4, 2022-12-08:
Yes, this was a small task where I learned something. :)
But still not enough.

> "The main function of touch is to interact with people."
> Sorry, I don't quite understand what you mean!

The question was:
What should I do with this information now.
How can I solve the touch problem now?

With this printf line I added, the board detects the touch point and position. 
The board seems to be a bit slower than before.

Now I can swipe left or right, but I had to keep my finger on the screen for a few seconds to change the desired action. 
The same goes for playing and pausing music.

So I added a number of milliseconds of delay(2000); to pause. This was too much and the board stopped responding at all! So I changed this to delay(200);

OK, next part; 
When I want to listen to the sound, i put on my sdcard,.. on my bluetooth speakers/headphones! How should I best proceed? do you have me advice ?

DeepCoreB4, 2022-12-11:
**Touch Fix!! This now works for me** 
    changed to this lines of code...
```
    ft3267_read_pos(&touch_points_num, &p.x, &p.y);
    if (p.x != 0 && p.y != 0) { // this is the fix!!!
      data->point.x = p.x;
      data->point.y = p.y;
    }
```
- Touch works now

closed 2022-12-11 by DeepCoreB4

euphi, 2023-01-02:
I created a library for easier use here https://github.com/fablabnbg/TRGBArduinoSupport