Xinyuan-LilyGO/LilyGo-T-RGB #10: Support for ESP-IDF

hayschan, 2022-12-03:
All of the current examples are based on Arduino framework for the Lilygo T-RGB. I wonder if there would be example for ESP-IDF. As the chip uses ESP32, it would be capable of running even more robust software through ESP-IDF and FreeRTOS.

mmMicky, 2022-12-12:
It supports esp-idf. If you use idf-v5.0 or above, it will even show that the quick record refresh will be faster and more stable.

krupis, 2023-07-30:
> I have newly added esp-idf support. https://github.com/Xinyuan-LilyGO/T-RGB/tree/main/example/esp-idf-example/T-RGB

Amazing stuff for preparing this project for the esp-idf. I have always thought the LilyGO products lack support for esp-idf!
I have tried to build it for the esp-idf but without any success. Perhaps you are familiar with this issue. I have created a seperate issue for this:
https://github.com/Xinyuan-LilyGO/T-RGB/issues/31

closed 2024-02-11 by github-actions[bot]