Xinyuan-LilyGO/LilyGo-T-RGB #11: Feature Request:  Integration with TFT-eSPI

PlastiBots, 2022-12-13:
Per the title, has anyone gotten this to work with TFT-eSPI, or any plans on it?  It's my go-to and I don't really want to have to try to figure out a new library when TFT-eSPI satisfies most of my needs.

mmMicky, 2022-12-14:
The name of TFT_eSPI indicates that it is only applicable to SPI interface. RGB interface is not supported.

elral, 2022-12-15:
Although it is called TFT_eSPI this library supports also 16Bit parallel access.
What this library makes very interesting, they have optimized driver for different processors, also for the esp32-s3. Optimisations are in regard of speed, and frame buffers can also be defined. Experiences with another display type shows that this speeds up the fps dramatically.
So it would be nice to have a display driver for this TFT_eSPI library. My assumption would be that the existing processor driver for the ESP32-S3 could still be used.

PlastiBots, 2022-12-15:
Yes, that is my understanding as well.  @mmMicky, I am very impressed with the devices as of late Lilygo is putting out.  However, the examples provided with this product have been nothing but trouble, and to date I cannot get anything to work properly.  I consider myself an intermediate to advanced user.  Many of us use TFT_eSPI, so it would be useful to have a familiar territory to go build upon.  I loath having to learn LVGL when I know I can get what I need done with the tools I already know.   Can we expect some TFT_eSPI examples?  I am overdue a review of this unit to @LilyGO, and I'd like to make it a positive one.  I am not sure if you work for them, or if they watch this, but suggest they should be paying attention if not.  We all want to help support your product.

mmMicky, 2022-12-16:
TFT_eSPI is a driver without RGB interface. I need to re-build with TFT_eSPI. There are so many variables in the library that I can't sort out the relationship. As a result, even if I complete the modification, there may be incompleteness.

closed 2022-12-20 by PlastiBots

jgauchia, 2023-01-16:
Hi, 
Maybe LovyanGFX library could work.
It's very similar to TFT_eSPI

DrJaymz, 2024-03-08:
I too fell down this hole today and it makes me very annoyed.  The TFT_eSPI is a work of art and its reason for existing is to extend GFX work.  Now, they have both evolved over time, but people sink significant effort into one or another and if you're like me you probably have created libs to do what you want and need to port them.

As I see it, TFT_eSPI can be made to work with this display because its all there.  I'm sure if someone sent BODMER a board it would be working by tea-time with an example commited by supper time.

My solution is currently to use TFT_eSPI library as-is on sprites and then push data from sprite to the screen, the performance is not too bad but there is probably some limitations I haven't yet come across.