Xinyuan-LilyGO/LilyGo-T-RGB #16: Issue: Using Grove Port for UART Comms and examples.

PlastiBots, 2022-12-30:
Earlier I posted code (#14) that demonstrated using the Grove port for UART purposes - it works. However, when I add that code into some of the examples provided, I get failures related to the (xl9535) which I believe relates to the use of Wire.begin creating an I2C connection on the same pins (see below) as the pins that are assigned to the Grove port which must be used for the Serial1 connection. I am trying to understand why the examples use Wire.begin in the Grove pins AND why it can't be removed/changed if I want to use UART instead?  There seems to be a tie-in to the xl9535 chip and code around it but I don't understand.  Is there another way to initiate Wire for the other components of T-RGB without using these 2 pins? (IIC_SDA_PIN, IIC_SCL_PIN)?  xl9535 fails.  
```Wire.begin(IIC_SDA_PIN, IIC_SCL_PIN, (uint32_t)400000);```

Here is an example sketch that fails.  Important, you have to have the SerialCAN adapter plugged into the Grove port to cause the failure.  To clarify the hardware is fine as I have it working per #14 as a base sketch. The below, brings in the screen bits and Serial CAN items.  Again, I think there's an issue with collision of Wire.begin being created on the same 2 pins.   What happens is once tft.init() triggers, it fails non-stop with "xl9535 not found".  Even though it works fine per #14 code. 

```
#include "FS.h"
#include <LittleFS.h>

#include "Wire.h"
#include "XL9535_driver.h"
#include "pin_config.h"
#include <Arduino.h>
#include <Arduino_GFX_Library.h>        // https://github.com/moononournation/Arduino_GFX.git
#include "ft3267.h"                     // Touch module
#include "Orbitron_ExtraBold12pt7b.h" 
#include "JpegFunc.h"                       //for displaying JPEGs using <JPEGDEC.h>  !!!Keep this at the bottom for all includes!
#include <Serial_CAN_Module.h> //for Serial CAN ODB unit
#include <HardwareSerial.h>    //for Serial CAN ODB unit

//****************************************************************************************
#define debug true    //true to turn debugging on
//****************************************************************************************

#define FORMAT_LITTLEFS_IF_FAILED false

HardwareSerial mySerial(1);    //Serial CAN ODB unit
Serial_CAN can;                //Serial CAN ODB unit

#define STANDARD_CAN_11BIT      1       // That depands on your car. some 1 some 0. 

#define PID_ENGIN_PRM       0x0C  //Serial CAN ODB unit
#define PID_VEHICLE_SPEED   0x0D  //Serial CAN ODB unit 
#define PID_COOLANT_TEMP    0x05  //Serial CAN ODB unit
#define PID_INTAKE_TEMP     0x0F  //Serial CAN ODB unit

#if STANDARD_CAN_11BIT
#define CAN_ID_PID          0x7DF   //noticed another sample had this: 0x7FC
#else
#define CAN_ID_PID          0x18db33f1
#endif

//Serial CAN ODB unit
int nSpeed = 0;         // speed, 0-200
int nRpm = 0;           // engin speed, 0-999
int nCTemp = 0;         // cooltant temp: 0-999
int nITemp = 0;         //intake temp: 0-999

//T-RGB
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    -1, -1, -1, EXAMPLE_PIN_NUM_DE, EXAMPLE_PIN_NUM_VSYNC, EXAMPLE_PIN_NUM_HSYNC, EXAMPLE_PIN_NUM_PCLK,
    EXAMPLE_PIN_NUM_DATA1, EXAMPLE_PIN_NUM_DATA2, EXAMPLE_PIN_NUM_DATA3, EXAMPLE_PIN_NUM_DATA4, EXAMPLE_PIN_NUM_DATA5,
    EXAMPLE_PIN_NUM_DATA6, EXAMPLE_PIN_NUM_DATA7, EXAMPLE_PIN_NUM_DATA8, EXAMPLE_PIN_NUM_DATA9, EXAMPLE_PIN_NUM_DATA10, EXAMPLE_PIN_NUM_DATA11, 
    EXAMPLE_PIN_NUM_DATA13, EXAMPLE_PIN_NUM_DATA14, EXAMPLE_PIN_NUM_DATA15, EXAMPLE_PIN_NUM_DATA16, EXAMPLE_PIN_NUM_DATA17);
Arduino_GFX *gfx = new Arduino_ST7701_RGBPanel(bus, GFX_NOT_DEFINED, 0 /* rotation */, false /* IPS */, 480, 480,
                                               st7701_type2_init_operations, sizeof(st7701_type2_init_operations), true,
                                               50, 1, 30, 20, 1, 30);

typedef struct {
  uint8_t cmd;
  uint8_t data[16];
  uint8_t databytes; // No of data in data; bit 7 = delay after set; 0xFF = end of cmds.
} lcd_init_cmd_t;

//for touch
typedef struct {
  uint16_t x;
  uint16_t y;
} tsPoint_t;

[... 653 lines cut ...]
    }
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found");
  }
}

void print_chip_info(void) {
  Serial.print("Chip: ");
  Serial.println(ESP.getChipModel());
  Serial.print("ChipRevision: ");
  Serial.println(ESP.getChipRevision());
  Serial.print("Psram size: ");
  Serial.print(ESP.getPsramSize() / 1024);
  Serial.println("KB");
  Serial.print("Flash size: ");
  Serial.print(ESP.getFlashChipSize() / 1024);
  Serial.println("KB");
  Serial.print("CPU frequency: ");
  Serial.print(ESP.getCpuFreqMHz());
  Serial.println("MHz");
}
*/

```

PlastiBots, 2023-01-08:
bump. Anyone have any ideas here?  Still having issues..  Unclear why the below Wire statement is included in the code? 
```Wire.begin(IIC_SDA_PIN, IIC_SCL_PIN, (uint32_t)400000);  ```

lewisxhe, 2023-05-22:
Change XL9555 from `Wire` to `Wire1`
Gove should initialize like this
```
const uint8_ t sda = 43;
const uint8_ t scl = 44;
Wire.begin(sda,scl)
```

s54b32julian, 2023-09-26:
@PlastiBots where you able to use the i2c port for an actual i2c device ? currently i can not get a device working on the i2c from the groove port. also it will not detected any i2c device i have with a i2c scan, am i missing something obviously config vise? Already tried to lower the speed to 100khz, but without success. I also dont understand why the i2c signal from Touch/XL is converted from 3V level to 3.3V, because all components should be 3.3V tolerant.

PlastiBots, 2023-09-26:
No.. I did not end up figuring it out..  It was a while back, but I gave up trying and moved onto another project. I vaguely recall it seeming like an issue with the Grove connection and some sort of conflict with Wire.begin and connecting the ODBII adapter via Grove. I suspected an issue with the config of the touch sensor driver on the I2C lines and how it's handled...  but not sure really.  Would love to hear of anyone else figured this out.

  Wire.begin(IIC_SDA_PIN, IIC_SCL_PIN, (uint32_t)400000);

s54b32julian, 2023-09-27:
@lewisxhe @LilyGO any idea how to add a i2c device or the problem i/we have?

extremerotary, 2023-10-10:
@s54b32julian 
I used this sketch and popped it on my Arduino MEGA - https://learn.adafruit.com/scanning-i2c-addresses/arduino
Then, I wired up the T-RGB I2C connector to the MEGA to the appropriate pins. 
The serial output from the MEGA looked like this - 
Scanning...
I2C device found at address 0x20  !
I2C device found at address 0x38  !
done

When I disconnect the SDA or SCL pins, the output is - 
Scanning...
No I2C devices found

To me, this shows that the T-RGB module is capable of communicating over I2C using that connector/pins. However, I too have been struggling to make these 2 modules communicate. It would be really nice if @LilyGO or @lewisxhe could provide a simple example sketch on I2C between two modules.

closed 2024-02-11 by github-actions[bot]

LasseKofoed, 2024-05-27:
Any one figured out how to communicate on the grove port with I2C. Seeing same thing as described.
With no cabel connecting this is my output.
23:25:31.107 > Scanning...
23:25:31.107 > I2C device found at address 0x20  !
23:25:32.108 > I2C device found at address 0x38  !
23:25:33.109 > done

Serial freezes when i plug in the grove cable with a I2C device on the other end.

Please  @LilyGO  @lewisxhe can you make a simple example on how to use the groveport for I2C communication.

extremerotary, 2024-05-27:
I never did figure it out and went wireless instead with BLE. However, I will say that I noticed LilyGo updated the repo and Readme and everything with a ton of additional info and changes in the past month. Maybe there is something in there that will help. Full transparency, I didnt read it all yet.