Xinyuan-LilyGO/LilyGo-T-RGB #17: SD-card not working

sandmen, 2022-12-31:
Hi,
I have tried several SD-cards.
Not one is working, or detected by the T-RGB !
Any hints?

PlastiBots, 2022-12-31:
I suggest this be merged with #13 that I created earlier - I'm having the same issue.

sandmen, 2023-01-02:
> I suggest this be merged with #13 that I created earlier - I'm having the same issue.

Workaround, you have to reduce the Clock for SPI.
I "just" set it to half, and the board detect SD-Cards.
There is already a patch around for the latest Arduino Framework.

mmMicky, 2023-01-03:
if (!SD_MMC.begin("/sdcard", true, true)) {
    Serial.println("Card Mount Failed");
    return;
  }

sandmen, 2023-01-03:
> 

Is for me not working... !
In SD_MMC.h 
SDMMC_FREQHIGHSPEED/2 is working...

have a look here
https://github.com/espressif/esp-idf/issues/8521

PlastiBots, 2023-01-07:
@sandmen, where is this variable?  I looked at SD_MMC.h in my version of the ESP 2.0.4 package (C:\Users\Dave\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\SD_MMC\src) and could not find an statement similar to the above.  SDMMC_FREQHIGHSPEED

sandmen, 2023-01-07:
@PlastiBots I'm using Plaformio and this uses the version 2.0.5...
In the SD_MMC.c file, should be somewhere something like this 
**host.max_freq_khz = sdmmc_frequency;**
Have a look for that, and check, where the sdmmc_frequency is come from !
You can do this as well
**host.max_freq_khz = sdmmc_frequency/2;**

PlastiBots, 2023-01-07:
Found it! Thx. Will give it a try.

lewisxhe, 2023-04-21:
Has the problem been solved ? If you still have problems, please reopen

closed 2023-04-21 by lewisxhe