Xinyuan-LilyGO/LilyGo-T-RGB #18: Support library for Arduino with platformio

euphi, 2023-01-02:
I noted that it's quite complicated to start with the T-RGB device. Example code is quite confusing (especially regarding touch) and you need to copy together a lot of things to get everything running.

Therefore I created a library that can be used with platformio with Arduino framework. See https://github.com/fablabnbg/TRGBArduinoSupport.

For now, touch is supported only for FT3267 ("Oval" version of T-RGB).

The library supports:

* initialisation of the display
* initialisation of the FT3267 touch controller
* integration of lvgl library (the library depends on it, so it is automatically installed and include-paths are set correctly).

It also comes with an example lvgl UI.

With our library, starting with T-RGB gets as easy as:

```C++
#include <TRGBSuppport.h>

TRGBSuppport trgb;
void setup() {
  trgb.init();
  // load your UI etc. (see example https://github.com/fablabnbg/TRGBArduinoSupport/tree/main/examples/ui_example)
}

void loop() {
  // your loop code
  // Handle lvgl timers
  lv_timer_handler();
  delay(2);
}

```

I would appreciate it, if you can add a link to the library in your README.MD

mmMicky, 2023-01-03:
A good idea.

lewisxhe, 2023-03-24:
Thank you for your contribution. I have added recommendations in README

closed 2023-03-24 by lewisxhe