Xinyuan-LilyGO/LilyGo-T-RGB #21: sketch too big and text section exceeds avilable space in board

tokio666, 2023-03-01:
Dear,

After some  changes in the sketch code and playing with different verisions of libraries I have compiled without code errors, but when it finish the message is... 

text section exceeds available space in board
Sketch too big

The board is configured as tells on the descriptions.

Any clue to fix this?

Sorry I am a newbie developer, but I expect that an example compile without issues

euphi, 2023-03-09:
You can configure larger partition for your app. 

See https://github.com/euphi/TRGB-BikeComputer/blob/main/partitions.csv and https://github.com/euphi/TRGB-BikeComputer/blob/main/platformio.ini for example.
(with the 16 MB flash of the T-RGB you can increase partition sizes even more than in this example)

BTW: See https://github.com/fablabnbg/TRGBArduinoSupport for how to  simplify creating and initializing a project with T-RGB.

lewisxhe, 2023-04-21:
Has the problem been solved ? If you still have problems, please reopen

closed 2023-04-21 by lewisxhe