Xinyuan-LilyGO/LilyGo-T-RGB #23: How to use GPIO0 for wakeup from deep sleep

euphi, 2023-03-09:
In many situations the touch interrupt can be to sensible for wakeup from deep sleep.

Unfortunately also no other GPIOs are available except GPIO0, so i tried to wakeup from deep sleep with GPIO0:

_from https://github.com/fablabnbg/TRGBArduinoSupport_:
```C++
void TRGBSuppport::deepSleep(void) {
  WiFi.disconnect();
  detachInterrupt(TP_INT_PIN);
  xl.pinMode8(0, 0xff, INPUT);
  xl.pinMode8(1, 0xff, INPUT);
  xl.read_all_reg();
  // If the SD card is initialized, it needs to be unmounted.
  if (SD_MMC.cardSize()) SD_MMC.end();

  digitalWrite(EXAMPLE_PIN_NUM_BK_LIGHT, EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL);

  Serial.println("Enter deep sleep");
  delay(4000);

  //esp_sleep_enable_ext0_wakeup((gpio_num_t)TP_INT_PIN, 0);		// Wakeup by touch
  pinMode(GPIO_NUM_0, INPUT_PULLUP);
  esp_sleep_enable_ext0_wakeup((gpio_num_t) GPIO_NUM_0, 0);	// Wakeup by boot-pin (TODO: not really useful for now, because display is not initialized correctly)
  esp_deep_sleep_start();
}

```

This works, the ESP also wakes up with GPIO0, but the display stays black after wakeup. Backlight seems to be switched on, but no displays output.
Touch seems to work fine, because I can see on serial that program logic reacts to touch input. (this includes changing the backlight brightness).

So what is the difference in wakeup between touch + interrupt and GPIO regarding the display? I changed nothing else and can reproduce correct behaviour (wakeup from `TP_INT_PIN`) and wrong behaviour (wakeup from `GPIO_NUM_0`) by just changing the one line with `esp_sleep_enable_ext0_wakeup()`.

I use your example code for TFT init:

```C++

void TRGBSuppport::tft_init(void) {
  xl.digitalWrite(LCD_CS_PIN, 1);
  xl.digitalWrite(LCD_SDA_PIN, 1);
  xl.digitalWrite(LCD_CLK_PIN, 1);

  // Reset the display and touch

  xl.digitalWrite(LCD_RST_PIN, 1);
  vTaskDelay(200 / portTICK_PERIOD_MS);
  xl.digitalWrite(LCD_RST_PIN, 0);
  xl.digitalWrite(TP_RES_PIN, 0);
  vTaskDelay(100 / portTICK_PERIOD_MS);
  xl.digitalWrite(LCD_RST_PIN, 1);
  xl.digitalWrite(TP_RES_PIN, 1);
  vTaskDelay(100 / portTICK_PERIOD_MS);

  ft3267_init(Wire);

  // Switch on backlight
  pinMode(EXAMPLE_PIN_NUM_BK_LIGHT, OUTPUT);
  digitalWrite(EXAMPLE_PIN_NUM_BK_LIGHT, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

  int cmd = 0;
  while (st_init_cmds[cmd].databytes != 0xff) {
    lcd_cmd(st_init_cmds[cmd].cmd);
    lcd_data(st_init_cmds[cmd].data, st_init_cmds[cmd].databytes & 0x1F);
    if (st_init_cmds[cmd].databytes & 0x80) {
      vTaskDelay(250 / portTICK_PERIOD_MS);
    }
    cmd++;
  }
  Serial.println("Register setup complete");
}

```

lewisxhe, 2023-04-13:
You can refer to here

https://github.com/Xinyuan-LilyGO/T-RGB/blob/3b8c7c82c80a197dbd2ac3e58ec20a74dd7ded56/examples/factory/factory.ino#L704

lewisxhe, 2023-04-21:
The problem is solved, I will close it, if there is still a problem, please reopen it

closed 2023-04-21 by lewisxhe

euphi, 2023-04-21:
It's not solved, because my problem is that display is not switched on after wakeup by GPIO0.

Wakeup itself is ok, but display backlight stays dark.

init / setup() is the same as for wakeup from touch interrupt.

BTW: I can't reopen the issue, because github doesn't allow it.

lewisxhe, 2023-04-22:
The factory example has been updated by me, please try again

DrJaymz, 2024-03-15:
I need a bit of help here - I wanted to sleep and then wake from a touch event.  I wake up the screen comes back on but is all sorts of messed up.  When waking from deep sleep I was under the impression that the board effectively resets and so the screen should be initialised as on power on.