Xinyuan-LilyGO/LilyGo-T-RGB #24: Board/Route view with resister designations shown for hardware hacking

slastrina, 2023-03-23:
Hi, im looking for a board view that shows the resistor designations i.e. RU1/2/3 and R4/5/6
Reason i ask is i have removed the sd card slot and wired an spi device to the old mosi, miso, cs and sck pins, i now need to remove and bridge appropriate pins to restore connectivity but its difficult to identify which resistor pads i need to short circuit to restore connectivity to the mcu. 

Are you able to assist.
Thanks

slastrina, 2023-04-03:
@LilyGO is the above request possible? an accompanying board layer routing diagram or a diagram indicating the resisters around the sd card

lewisxhe, 2023-04-03:
[image]
[image]

I hope it can help you

slastrina, 2023-04-04:
Hi, yes that's a big help, though any chance you can provide the bottom layer as well, just so I can confirm where the vias are going

lewisxhe, 2023-04-04:
I will provide it to you in two days, and I am currently on vacation

slastrina, 2023-04-04:
No rush, thanks for your help, enjoy your holiday

lewisxhe, 2023-04-07:
[image]

slastrina, 2023-04-08:
Thats a big help, really appreciate it

closed 2023-04-08 by slastrina