Xinyuan-LilyGO/LilyGo-T-RGB #26: Sample does not compile for 2.1" H597

themagicm, 2023-04-13:
Using PlatformIO and VSCode, error compiling:

 `Compiling .pio/build/esp32s3box/libef7/GFX Library for Arduino/display/Arduino_HX8369A.cpp.o
Compiling .pio/build/esp32s3box/libef7/GFX Library for Arduino/display/Arduino_ILI9225.cpp.o
src/main.cpp:19:123: error: no matching function for call to 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int)'
     EXAMPLE_PIN_NUM_DATA13, EXAMPLE_PIN_NUM_DATA14, EXAMPLE_PIN_NUM_DATA15, EXAMPLE_PIN_NUM_DATA16, EXAMPLE_PIN_NUM_DATA17);
                                                                                                                           ^
In file included from .pio/libdeps/esp32s3box/GFX Library for Arduino/src/Arduino_GFX_Library.h:14,
                 from src/main.cpp:7:
.pio/libdeps/esp32s3box/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h:59:3: note: candidate: 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, int32_t, bool, uint16_t, uint16_t)'
   Arduino_ESP32RGBPanel(
   ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32s3box/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h:59:3: note:   candidate expects 33 arguments, 23 provided
.pio/libdeps/esp32s3box/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h:56:7: note: candidate: 'constexpr Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(const Arduino_ESP32RGBPanel&)'
 class Arduino_ESP32RGBPanel
       ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32s3box/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h:56:7: note:   candidate expects 1 argument, 23 provided
.pio/libdeps/esp32s3box/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h:56:7: note: candidate: 'constexpr Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(Arduino_ESP32RGBPanel&&)'
.pio/libdeps/esp32s3box/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h:56:7: note:   candidate expects 1 argument, 23 provided
src/main.cpp:22:24: error: expected type-specifier before 'Arduino_ST7701_RGBPanel'
 Arduino_GFX *gfx = new Arduino_ST7701_RGBPanel(bus, GFX_NOT_DEFINED, 0 /* rotation */, false /* IPS */, 480, 480,
                        ^~~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/esp32s3box/src/main.cpp.o] Error 1`

lewisxhe, 2023-04-14:
I have no problem compiling, please try to clone the warehouse again
[image]

themagicm, 2023-04-14:
aaah I see..  Arduino GFX 1.3.0.  It doesnt compile with GFX For Arduino by the same dev, version 1.3.4.

PlastiBots, 2023-04-14:
I came across this as well. You need to change the way you call Arduino_ESP32RGBPanel and Arduino_DataBus.  Look at the [sample ](https://github.com/moononournation/Arduino_GFX/blob/master/examples/PDQgraphicstest/PDQgraphicstest.ino)in Arduino_GFX to see the updated way.

Here's an example that works for my T-RGB:

```

//per Arduino_GFX 1.3.1: USE THIS
//Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
//    -1, -1, -1, EXAMPLE_PIN_NUM_DE, EXAMPLE_PIN_NUM_VSYNC, EXAMPLE_PIN_NUM_HSYNC, EXAMPLE_PIN_NUM_PCLK,
//    EXAMPLE_PIN_NUM_DATA1, EXAMPLE_PIN_NUM_DATA2, EXAMPLE_PIN_NUM_DATA3, EXAMPLE_PIN_NUM_DATA4, EXAMPLE_PIN_NUM_DATA5,
//    EXAMPLE_PIN_NUM_DATA6, EXAMPLE_PIN_NUM_DATA7, EXAMPLE_PIN_NUM_DATA8, EXAMPLE_PIN_NUM_DATA9, EXAMPLE_PIN_NUM_DATA10, EXAMPLE_PIN_NUM_DATA11,
//    EXAMPLE_PIN_NUM_DATA13, EXAMPLE_PIN_NUM_DATA14, EXAMPLE_PIN_NUM_DATA15, EXAMPLE_PIN_NUM_DATA16, EXAMPLE_PIN_NUM_DATA17);

//Arduino_GFX *gfx = new Arduino_ST7701_RGBPanel(bus, GFX_NOT_DEFINED, 0 /* rotation */, false /* IPS */, 480, 480,
//                                               st7701_type2_init_operations, sizeof(st7701_type2_init_operations), true,
//                                               50, 1, 30, 20, 1, 30);
//END USE THIS Arduino_GFX  1.3.1

//USE THIS per Arduino_GFX  v 1.3.2:  sample: look for T-RGB: https://github.com/moononournation/Arduino_GFX/blob/master/examples/PDQgraphicstest/PDQgraphicstest.ino
Arduino_DataBus *bus = new Arduino_XL9535SWSPI(8 /* SDA */, 48 /* SCL */, 2 /* XL PWD */, 3 /* XL CS */, 5 /* XL SCK */, 4 /* XL MOSI */);

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  45 /* DE */, 41 /* VSYNC */, 47 /* HSYNC */, 42 /* PCLK */,
  21 /* R0 */, 18 /* R1 */, 17 /* R2 */, 16 /* R3 */, 15 /* R4 */,
  14 /* G0 */, 13 /* G1 */, 12 /* G2 */, 11 /* G3 */, 10 /* G4 */, 9 /* G5 */,
  7 /* B0 */, 6 /* B1 */, 5 /* B2 */, 3 /* B3 */, 2 /* B4 */,
  1 /* hsync_polarity */, 50 /* hsync_front_porch */, 1 /* hsync_pulse_width */, 30 /* hsync_back_porch */,
  1 /* vsync_polarity */, 20 /* vsync_front_porch */, 1 /* vsync_pulse_width */, 30 /* vsync_back_porch */,
  1 /* pclk_active_neg */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
  480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
  bus, GFX_NOT_DEFINED /* RST */, st7701_type4_init_operations, sizeof(st7701_type4_init_operations));
//END USE THIS Arduino_GFX v1.3.2
```

lewisxhe, 2023-04-21:
The problem is solved, I will close it, if there is still a problem, please reopen it

closed 2023-04-21 by lewisxhe