Xinyuan-LilyGO/LilyGo-T-RGB #27: How to create PlatformIO Partition file to use all 16MB PSRAM

PlastiBots, 2023-05-18:
I have a project storing a number of large images directly in PSRAM (not using SPIFFS etc). I am currently using the _max_app_8MB.csv_ partition setting in my platformio.ini as the _default_16MB.csv_ does not work.   I have seen the options linked below but am hoping someone can advise on the right partition table configuration that allocates all 16MB PSRAM for internal memory storage (not SPIFFS or LittleFS etc). 

[ https://github.com/euphi/TRGB-BikeComputer/blob/main/partitions.csv]( https://github.com/euphi/TRGB-BikeComputer/blob/main/partitions.csv) and [https://github.com/euphi/TRGB-BikeComputer/blob/main/platformio.ini](https://github.com/euphi/TRGB-BikeComputer/blob/main/platformio.ini) for example.

Here's my current working config (with 8MB):
`
[env:esp32-s3-trgb]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
lib_deps = 
	moononournation/GFX Library for Arduino@^1.3.4
	fbiego/ESP32Time@^2.0.0
	northernwidget/DS3231@^1.1.2
upload_speed = 921600
;build_flags = -DBOARD_HAS_PSRAM
build_flags = 
	;-DARDUINO_USB_MODE=1  ;with this removed, device auto reboots after flashing (good)  0=hardware USB-CDC or 1=TinyUSB
	-DARDUINO_USB_CDC_ON_BOOT=1  ;allows serial monitor to work  must also have monitor_speed set
	-DBOARD_HAS_PSRAM

board_build.arduino.memory_type = qio_opi
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = max_app_8MB.csv
`

lewisxhe, 2023-05-19:
Firstly, T-RGB only has PSRAM of 8MB bytes, while Flash only has 16MB bytes If you want to change the partition table, please check the details and instructions here [Partition Tables](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#:~:text=The%20partition%20table%20which%20is%20flashed%20to%20the,used%20to%20convert%20between%20CSV%20and%20binary%20formats.)

PlastiBots, 2023-05-19:
Thanks for the clarification and link.  However, I'm new at this bit and don't understand what needs to be done.  Does anyone have a sample file they can share that uses the full 16MB flash?

lewisxhe, 2023-05-19:
As esp said, if you want to maximize the use of flash, then directly select the partition table of <Single factory app, no OTA>, and add `board_build.partitions = app16M.csv` in <platformio.ini>

Then create the <app16M.csv> file in the same directory as <platformio.ini>, and fill in the following example

```
# Name,   Type, SubType, Offset,  Size,   Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 15M,
```

lewisxhe, 2023-05-19:
Oh... I now know what your problem is. Please modify the board to

board = esp32s3box

PlastiBots, 2023-05-20:
Yes! That works.  Using board = esp32s3box and the partition table you provided above solved it.  Thanks for helping figure this out.

closed 2023-05-20 by PlastiBots