Xinyuan-LilyGO/LilyGo-T-RGB #29: What pin for buzzer can I use?

karfly, 2023-07-15:
Hi!
I want to attach buzzer to Lilygo T-RGB chip. What pin (from [here](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/examples/factory/pin_config.h)) can I use?
I want to send some tone() to it.

karfly, 2023-07-15:
@lewisxhe @LilyGO @euphi @slastrina
Guys, maybe you can help

lewisxhe, 2023-07-16:
See doc https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/ledc.html

karfly, 2023-07-16:
@lewisxhe thank you!
I'm trying to use GPIO37 (according to [schematic](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/schematic/T-RGB.pdf) it's not connected to anything).

I set `OUTPUT` pin mode for 37 pin:
```c
Serial.println("buzzer start");
pinMode(37, OUTPUT);
Serial.println("buzzer end");
```

And chip starts to restart all the time with such log:
```log
buzzer start
buzzer end
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037795d
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
...
```

How can I properly use this pin for buzzer?

lewisxhe, 2023-07-16:
33-37 is not available and is already occupied by PSRAM.

karfly, 2023-07-16:
@lewisxhe I see, thank you.
Is it possible to use GPIO38 then? If I'm not going to utilize SD card?

karfly, 2023-07-16:
Actually I tried it with GPIO38 and it works but with some issues. I have a standard 2-wire piezo buzzer (GND wire and Signal wire). I attached GND wire to chip's GND and Signal wire to GPIO38.

Here're the issues:
1. When buzzer is physically attached to GPIO38 – touch screen is not working. When it's not attached physically (but code for playing buzzer tones is running) – touch screen successfully working.
2. There is some clapping noise I hear through buzzer all the time.

lewisxhe, 2023-07-18:
IO38 has been used by the SD card, and an external pull-up resistor is added, so this is the source of the noise, if you don't need the SD card function, then you can remove the resistor
[image]

lewisxhe, 2023-07-18:
IO38 is not used by touch, so it should not affect touch functionality

karfly, 2023-07-18:
@lewisxhe thank you very much!
I successfully attached buzzer to GPIO40 and works great ❤️ 

Can you share the schematic from the screenshot above?

closed 2024-02-10 by github-actions[bot]