Xinyuan-LilyGO/LilyGo-T-RGB #32: T-Display-RGB issues running sample esp-idf project

krupis, 2023-07-31:
Hello. I have recently ordered T-Display-RGB development board and flashed the sample project : 
https://github.com/Xinyuan-LilyGO/T-RGB/tree/main/examples/esp-idf-example/T-RGB

After experimenting with this sample project for a while, I have found 3 bugs and would like to know whether its due to faulty hardware or some other issue:

**Issue1**

I have noticed that sometimes when I start the board, it is detecting touches at random positions such as : ```[0] point x: 4095  point y: 4095 ```

I cant wrap my head around what could cause my board to detect touch even when I am not touching the display at all. The only thing I can think of is faulty touch panel.

[1B][0;32mI (880) esp_psram: SPI SRAM memory test OK[1B][0m
[1B][0;32mI (889) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (889) cpu_start: cpu freq: 160000000 Hz[1B][0m
[1B][0;32mI (889) cpu_start: Application information:[1B][0m
[1B][0;32mI (892) cpu_start: Project name:     T-RGB[1B][0m
[1B][0;32mI (897) cpu_start: App version:      1[1B][0m
[1B][0;32mI (901) cpu_start: Compile time:     Jul 31 2023 17:35:05[1B][0m
[1B][0;32mI (907) cpu_start: ELF file SHA256:  6751df4f44e82a56...[1B][0m
[1B][0;32mI (913) cpu_start: ESP-IDF:          v5.0.1[1B][0m
[1B][0;32mI (918) cpu_start: Min chip rev:     v0.0[1B][0m
[1B][0;32mI (923) cpu_start: Max chip rev:     v0.99 [1B][0m
[1B][0;32mI (928) cpu_start: Chip rev:         v0.1[1B][0m
[1B][0;32mI (932) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (940) heap_init: At 3FC98A58 len 00050CB8 (323 KiB): D/IRAM[1B][0m
[1B][0;32mI (946) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM[1B][0m
[1B][0;32mI (953) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[1B][0m
[1B][0;32mI (959) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM[1B][0m
[1B][0;32mI (965) esp_psram: Adding pool of 7296K of PSRAM memory to heap allocator[1B][0m
[1B][0;32mI (973) spi_flash: detected chip: winbond[1B][0m
[1B][0;32mI (977) spi_flash: flash io: dio[1B][0m
[1B][0;33mW (981) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.[1B][0m
[1B][0;32mI (995) cpu_start: Starting scheduler on PRO CPU.[1B][0m
[1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
[1B][0;32mI (1005) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations[1B][0m
[1B][0;32mI (1015) T-RGB example: Initialize xl9535[1B][0m
[1B][0;32mI (1025) T-RGB example: Initialize touch[1B][0m
43:11: [1B][0;32mI (1425) T-RGB example: Initialize st7701s[1B][0m
43:14: [1B][0;32mI (4215) T-RGB example: Turn off LCD backlight[1B][0m
[1B][0;32mI (4215) T-RGB example: Install RGB LCD panel driver[1B][0m
[1B][0;32mI (4225) T-RGB example: Register event callbacks[1B][0m
[1B][0;32mI (4225) T-RGB example: Initialize RGB LCD panel[1B][0m
[1B][0;32mI (4225) T-RGB example: Turn on LCD backlight[1B][0m
[1B][0;32mI (4225) T-RGB example: Initialize LVGL library[1B][0m
[1B][0;32mI (4235) T-RGB example: Allocate separate LVGL draw buffers from PSRAM[1B][0m
[1B][0;32mI (4245) T-RGB example: Register display driver to LVGL[1B][0m
[1B][0;32mI (4245) T-RGB example: Register touch driver to LVGL[1B][0m
[1B][0;32mI (4255) T-RGB example: Install LVGL tick timer[1B][0m
[1B][0;32mI (4255) T-RGB example: Test decoding pictures.[1B][0m
43:15: getPointNum: 1  

[0] point x: 178  point y: 245 

getPointNum: 1  

[0] point x: 178  point y: 245 

getPointNum: 1  

[0] point x: 178  point y: 245 

getPointNum: 1  

[0] point x: 178  point y: 245 

getPointNum: 1  

[0] point x: 178  point y: 245 

getPointNum: 1  
[... 299 lines cut ...]
  LV_IMG_DECLARE(logo);
  ESP_LOGI(TAG, "Test decoding pictures.");
  lv_obj_t *img = lv_img_create(lv_scr_act());
  lv_img_set_src(img, &logo);
  lv_obj_center(img);
```

Logo image displayed in yellow:

[image]

After a couple of reboots, it displays properly:
[image]

**Issue4**

I have played around with T-Display-S3 ST7789 development board. For the project that I was working on I had to display color gradients and I found it difficult to display complex color gradients since st7789 only supports 16 bit color depth. (RGB565).

I was hoping to avoid any issues with the T-Display-RGB since I was under impression it supports 24 bit color depth (RGB565).

In the menuconfig, I can only enable 32 bit color depth such as:
[image]
Is that the same thing as 24 bit color depth and is it safe to use?

I would very much appreciate help finding answer to these 4 issues

PlastiBots, 2023-07-31:
I recall having a similar issue a while back.  Check the SPI bus speed parameter you are using to setup the touch.  I recall having to change it from the default. Im not in front of my computer right now, so can check later.

PlastiBots, 2023-08-01:
This>>  Wire.begin(IIC_SDA_PIN, IIC_SCL_PIN, (uint32_t)400000);

krupis, 2023-08-03:
@PlastiBots 
I am working with esp-idf and not Arduino framework. 

Either way, I have tried to reduce the I2C clock frequency but I am able to reproduce all of the above issues on 2 seperate T-Display RGB development boards.

Hopefully someone from LiLyGo can help debug the issues

closed 2024-02-10 by github-actions[bot]