Xinyuan-LilyGO/LilyGo-T-RGB #34: How to use 24-bit color depth mode

krupis, 2023-08-07:
Hello. I have recently got 2.1 inch full circle display form Lilygo:
https://www.lilygo.cc/products/t-rgb

One of the main reasons why I have decided to try this display because I thought it supports 24 bit color depth (RGB888) mode.
I have been trying to test 24 bit color depth using LVGL v9 (latest master branch but not officially released yet):
https://github.com/lvgl/lvgl

I have used esp-idf example code provided by Lilygo:
https://github.com/Xinyuan-LilyGO/T-RGB/tree/main/examples/esp-idf-example/T-RGB

I just stripped it down by removing unnecessary functions and changed the LVGL api slightly to make sure it compiles without any issues with LVGL v9. You can find my project source code here:
https://github.com/krupis/Round_RGB
It should build and run without any issues with esp-idf v5.0.3

I am trying to get the 24 bit color mode working. I have only made a couple of code modifications:

**Modification1**
Change color depth to 24 (RGB888) in menuconfig:
[image]

**Modification2**
Replace one line in st_init_cmds that sets the COLMOD register according to ST7701 documentation:
https://focuslcds.com/content/ST7701.pdf

[image]

```
{0x3a, {0x66}, 0x01},
```
replaced with:
```
{0x3a, {0x70}, 0x01},
```

Despite my efforts, I cannot get the 24 bit color depth mode to work properly as the display is displaying weird things when trying to display LVGL pie chart:
[image]

I would really appreciate help getting 24 bit color depth working on this display to use it to its full potential.
Thanks in advance!

Additionally, I would like to understand why the following command COLMOD command is sent by default:
```
{0x3a, {0x66}, 0x01}
```

According to ST7701 documentation, it should set the display to 18 bit color mode. Why would that be the case and why it works with LVGL set to 16 bit color depth?

lewisxhe, 2023-08-17:
ESP32S3 only supports RGB565 mode

See ,  https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/api-reference/peripherals/lcd.html

[image]

As for why 0X3A is set to 666, it is because the screen itself only supports RGB666 mode.

krupis, 2023-08-17:
@lewisxhe 
Thanks for response.

I will contact esp-idf support directly regarding 24 bit color depth support.

As per the screen itself, the st7701 documentation suggests that it supports RGB888. I was under impression that the T-Display-RGB displatys are ST7701 based.

Can you clarify this to me please?

lewisxhe, 2023-08-17:
Yes, T-RGB is based on st7701, but this is only a driver chip. Screen manufacturers only support RGB666 mode, and sending other instructions is invalid. This is an internal limitation of the screen and cannot be changed through software.

lewisxhe, 2023-08-17:
RGB888 requires hardware connections of 24 GPIOs, while TRGB only connects 16 data cables. All other changes are invalid.

krupis, 2023-08-17:
@lewisxhe

Oh fair enough... Didint realize that..

How can I know which particular screen is used? Is there a possibility to mount another screen, one that would support rgb888?

In short, what would be the easiest way to have a display that would support rgb888?

Perhaps you have any plans on releasing such product sometime soon? It would be really cool. Sometimes when I need to display complex gradients, rgb565 or rgb666 wont do it

lewisxhe, 2023-08-17:
Currently, there is no screen that supports RGB888, as ESP only supports 565

closed 2024-02-10 by github-actions[bot]