Xinyuan-LilyGO/LilyGo-T-RGB #37: ESP-IDF flash issue

rightcornerengineering, 2023-10-09:
I've tried to flash the example for IDF.
My IDF version is 5.0.1. I was able to successfully build by `idf.py build`
And when I tried `idf.py flash monitor`, I got error as following:
`A fatal error occurred: Packet content transfer stopped (received 25 bytes)`

I've also tried just `hello world` example to flash and got the same error as above. It seems device issue? But I could do `idf.py erase_flash` and it works. Or something wrong with `esptool.py`?

lewisxhe, 2023-10-13:
Are you using win?

I write normally using wsl. If you are using win, I remember that you need to select a USB-jatg board to write.

[image]

rightcornerengineering, 2023-10-13:
> Are you using win?
> 
> I write normally using wsl. If you are using win, I remember that you need to select a USB-jatg board to write.
> 
> [image]

I am using Linux. I verified port number is present. And also was able to do 'erase-flash' and 'monitor'. Only cannot 'flash'.

lewisxhe, 2023-10-13:
Have you tried adding the -b 115200 parameter?

rightcornerengineering, 2023-10-13:
> Have you tried adding the -b 115200 parameter?

Yes I tried that too to lower the baud rate.

lewisxhe, 2023-10-13:
I have encountered this problem before, please try again
1. Press and hold the BOOT button and keep pressing
2. Press RST and release
3. Release the BOOT button,
Try using `idf.py -b 115200 flash`

rightcornerengineering, 2023-10-13:
> I have encountered this problem before, please try again
> 
> 1. Press and hold the BOOT button and keep pressing
> 2. Press RST and release
> 3. Release the BOOT button,
>    Try using `idf.py -b 115200 flash`

Trust me. I have already tried this steps. Still no luck.
I also shared a video with you about following the steps.

rightcornerengineering, 2023-10-13:
@lewisxhe 
`idf.py erase-flash` works to clean up.
`idf.py flash` does not work.
`idf.py monitor` works to see messages but it keeps restarting with `E (67) boot: No bootable app partitions in the partition table`.

I think the PCB is defective on flash memory side.

lewisxhe, 2023-10-13:
One last try, use ESP DOWNLOAD TOOL for testing. .

https://github.com/Xinyuan-LilyGO/T-RGB/tree/main/firmware

rightcornerengineering, 2023-10-13:
> One last try, use ESP DOWNLOAD TOOL for testing. .
> 
> https://github.com/Xinyuan-LilyGO/T-RGB/tree/main/firmware

Ok so I flashed the firmware `[TRGB_GT911_2_8_Inc.bin](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/firmware/TRGB_GT911_2_8_Inc.bin)` and got the following message
```
E (50) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (50) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[   508][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
```

rightcornerengineering, 2023-10-13:
> Ok so I flashed the firmware `[TRGB_GT911_2_8_Inc.bin](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/firmware/TRGB_GT911_2_8_Inc.bin)` and got the following message
>
> ```
> E (50) sdmmc_common: [...]

However, after this, I also tried to flash using a command `idf.py flash` and got still the same error. There must be something wrong on flash module.

lewisxhe, 2023-10-13:
If the esp download tool can write successfully, then the board is fine.

You said there is a problem with the flash memory, so it is impossible to write successfully through the esp download tool.

As for why idf cannot be written, I don't know.
Maybe you should open an issue in the esp-idf community

lewisxhe, 2023-10-13:
Maybe you should try using the following command to test write,

```esptool.py -b 115200 --no-stub -p /dev/ttyS49 write_flash 0x00 ./build/bootloader/bootloader.bin```

I found that if I don’t add --no-stub, I won’t be able to write. I have to enter BOOT mode before I can use it.

lewisxhe, 2023-10-13:
[image]
Maybe you should try checking this off.

rightcornerengineering, 2023-10-13:
> [image] Maybe you should try checking this off.

This has been off.
And I found out that even flashing using esptool is not always successful. Sometimes fails still.
And I have many esp32 and esp32s3 and they never had such problem but only Lilygo product. It must be some issues on hardware.

lewisxhe, 2023-10-14:
Have you tried this before?
```
esptool.py -b 115200 --no-stub -p /dev/ttyS49 write_flash 0x00 ./build/bootloader/bootloader.bin
```

rightcornerengineering, 2023-10-16:
> Have you tried this before?
> 
> ```
> esptool.py -b 115200 --no-stub -p /dev/ttyS49 write_flash 0x00 ./build/bootloader/bootloader.bin
> ```

Isn't it only for bootloader firmware? What about application and other bins?

lewisxhe, 2023-10-17:
What I mean is whether it is possible to burn the program by calling `esptool.py`,
For example, `idf.py build` will output the burning command (as shown below). You can directly copy this command and add the two parameters `-b 115200 --no-stub` and try to see if it can be written normally.

[image]

rightcornerengineering, 2023-10-17:
> What I mean is whether it is possible to burn the program by calling `esptool.py`,
>
> For example, `idf.py build` will output the burning command (as shown below). You can directly copy this command [...]

Ok so this way it can flash through which is good. But it is still the problem where `idf.py flash` keeps giving error.

rightcornerengineering, 2023-10-17:
And I've flashed with LVGL demo as per the description of this source code with double buffer enabled in `menuconfig`.
It flashed but the display is flickering when especially it's on `Analytics` section and with graphs moving. Is it known issue or is there a way to fix this?

lewisxhe, 2023-10-18:
I also encountered the same problem, please watch my video

You need to manually put esp32 into download mode before you can write to flash.

https://youtu.be/9ZamjdNmxao

rightcornerengineering, 2023-10-18:
> I also encountered the same problem, please watch my video
> 
> You need to manually put esp32 into download mode before you can write to flash.
> 
> https://youtu.be/9ZamjdNmxao

Your issue is different from my issue. I shared a video with your colleague before. Could you check and see the error message from mine?

lewisxhe, 2023-10-19:
I suggest you update idf to a newer version. Through searching, I found someone who encountered the same situation as you.
https://github.com/espressif/esptool/issues/857

Theoretically speaking, as long as one tool can be programmed normally, then the hardware is normal. If it cannot be written, then there is a problem.

rightcornerengineering, 2023-10-19:
> And I've flashed with LVGL demo as per the description of this source code with double buffer enabled in `menuconfig`. It flashed but the display is flickering when especially it's on `Analytics` section and with graphs moving. Is it known issue or is there a way to fix this?

@lewisxhe 
Could you check if you have seen this before? I was wondering if this may be related to hardware.

lewisxhe, 2023-10-19:
I don't think it's a hardware problem, otherwise esptool.py wouldn't have worked

closed 2024-02-10 by github-actions[bot]