Xinyuan-LilyGO/LilyGo-T-RGB #41: Serial.print does not work in "Factory" example

ThomasSlowhand, 2024-01-09:
Help/advice needed
I am using the T-RGB example "factory" (this app which is commonly installed on a T-RGB at supply), as a framework for creating a new app, but the "Serial.print" debug function does not work, even if "Serial.begin(115200)" is declared in the beginning of "void setup()". normally. This is not an installation fault, as the Serial.print function works with other apps on the same device. Any help is highly appreciated.

lewisxhe, 2024-01-09:
[image]
CDC must turn on

ThomasSlowhand, 2024-01-09:
Hello Lewis,

many thanks, problem solved!

BR,
Thomas

closed 2024-01-29 by lewisxhe