Xinyuan-LilyGO/LilyGo-T-RGB #42: Debug option aside "Serial.print" with PlatformIO

ThomasSlowhand, 2024-01-10:
Is there a solution for a professional debug option to be used except "Serial.print" with PlatformIO?

lewisxhe, 2024-01-10:
See [here](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/platformio.ini#L34-L34)

ThomasSlowhand, 2024-01-10:
# @lewisxhe 
hello Lewis,
many thanks for your hint, but that is only for serial debug via USB. I am asking for a professional debug solution which allows f. e. step-by-step debug, and real time value inspection, for example a JTAG solution.

lewisxhe, 2024-01-10:
I haven't used USB-Jata, maybe you should look at this 
https://github.com/platformio/platform-espressif32/issues/828

ThomasSlowhand, 2024-01-11:
Hello Lewis,

many thanks, OpenOCD works with PIO, got T-RGB 2.1" fullcircle running with these entries in platformio.ini of the project:

[env]
platform = espressif32
framework = arduino
platform_packages = 
	tool-openocd
monitor_speed = 115200
upload_speed = 921600
debug_tool = esp-builtin
debug_init_break = tbreak setup
debug_speed = 500
build_flags = 
	-DBOARD_HAS_PSRAM
	-DLV_LVGL_H_INCLUDE_SIMPLE
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1

No other actions to be taken.

But there are still issues with a GDB timeout which is set too short, did not find out yet how and where to set this value in PlatformIO:

>> [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (1085 ms). Workaround: increase "set remotetimeout" in GDB<<

lewisxhe, 2024-01-12:
As I said above, I haven't used the esp built-in JTAG, so I can't help you either

ThomasSlowhand, 2024-01-12:
I just wanted to let you know, that in case of another customer asks for the same issue, you could support him by giving him the hint to use openOCD.

ThomasSlowhand, 2024-01-18:
Hello Lewis,

 

can you fix this, as the lib is an remix of your “factory” example?

 

Mit freundlichen Grüßen,

best regards

 

Thomas Krim

lewisxhe, 2024-01-19:
What do you need me to do?

ThomasSlowhand, 2024-01-19:
Sorry, wrong issue. This one is closed. Please see my other message regarding TRGBArduinosupport lib.

closed 2024-01-19 by ThomasSlowhand