Xinyuan-LilyGO/LilyGo-T-RGB #43: TRGBArduinosupport lib does not work properly

ThomasSlowhand, 2024-01-16:
You recommended:

>> Easy to use RGB panel library, organized by [fablabnbg/TRGBArduinoSupport](https://github.com/fablabnbg/TRGBArduinoSupport),This is more recommended for novices<<

**TRGBArduinosupport** works fine with FT3267 board, but with CST820 board the touch-function raises a fault in PIO serial monitor:

Connecte[ 15731][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[ 15761][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[ 15791][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1

With GT911 boards the touch-function does not work  at all.

lewisxhe, 2024-01-19:
Because the CST820 enters sleep mode, the CST820 will not respond from I2C until the touch starts.

ThomasSlowhand, 2024-01-19:
Hello Lewis,

with CST820 the touch is signaled, but is not properly handled by TRGBArduinosupport, this error msg is raised with every touch:

Connecte[ 15731][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[ 15761][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[ 15791][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1

lewisxhe, 2024-01-19:
You shouldn't get errors when touching. Can touch be used normally except for reporting errors?

ThomasSlowhand, 2024-01-19:
No, CST820 and GT911 the touch function is faulty and does not work, touch is not recognized by the software using TRGBArduinosupport.. But only CST820 generates the mentioned error msg in the serial monitor. Because this lib is a re-mixed excerpt from your "factory" example I hope you might have more clue about this issue.

lewisxhe, 2024-01-19:
TRGBArduinoSupport is written by fablabnbg. He has stated in the [README](https://github.com/fablabnbg/TRGBArduinoSupport?tab=readme-ov-file#touchpad) that he only uses FT3267 and does not support other models. If you need to use other models. If you use other two types of touch, please use the examples provided by this warehouse, or modify TRGBArduinoSupport.

ThomasSlowhand, 2024-01-19:
He has updated the lib in 09/2023, please read
[https://github.com/fablabnbg/TRGBArduinoSupport/issues/5](url),
but it not working properly as already reported.

ThomasSlowhand, 2024-01-19:
[https://github.com/fablabnbg/TRGBArduinoSupport/commit/7d47e336f9bb236658f8b0510bc24568eb030d91](url)

lewisxhe, 2024-01-19:
T-RGB was maintained by @miky in the early days. It is really not good for novices. I have been integrating the LilyGo display series recently and TRGB is also planned. Please give me some time and I will reorganize T-RGB.

ThomasSlowhand, 2024-01-19:
Hello Lewis,

T-RGB is a really good product, and is very useful for my project for which I am planning to sell a few hundred items if I can proceed with it. I already have on project ready with 2.1" halfcircle, but its screen shape is too weird and it is to complicated to mount, because it needs a wide bezel to fasten which looks ugly. I therefore definitely need 2.1" fullcircle & 2.8", but as long as I cannot use the TRGBArduinosupport with it, both my projects are halted.I do not know what you mean with "I will reorganize T-RGB.", my only need is to get TRGBArduinosupport working with these two board versions.

lewisxhe, 2024-01-19:
I understand, what I mean is to integrate the drivers together, and there is no need to distinguish the driver model. I have already implemented many cases, and TRGB is not difficult.  Please give me a few days.

lewisxhe, 2024-01-25:
@ThomasSlowhand 
I have completed the upgrade to T-RGB and now it is installed in the form of a library. Users do not need to distinguish between 2.1 and 2.8, nor what touch model it is.

lewisxhe, 2024-01-25:
I will register the platformio and arduino libraries for T-RGB soon. Then you can install it directly from pio or arduino. Please use it manually now.
```
lib_deps =
https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB
```

ThomasSlowhand, 2024-01-25:
Hello Lewi,

 

many thanks for your message, these are very good news. I will check it immediately.

 

Mit freundlichen Grüßen,

best regards

 

Thomas Krim

ThomasSlowhand, 2024-01-25:
Many thanks again, Lewis, I will keep you posted about my efforts. Do the env switches in platformio.ini and the defines regarding the used version of the boards have all to be removed?

Mit freundlichen Grüßen,
best regards

Thomas Krim

https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB

lewisxhe, 2024-01-25:
Below is my demo platformio.ini. You can develop directly in the root directory of LilyGo-T-RGB, or you can integrate LilyGo-T-RGB into your project.
```
[env:T-RGB]
platform = espressif32@6.3.0
board = esp32s3box
framework = arduino
board_build.partitions = default_16MB.csv
upload_speed =  921600
monitor_speed = 115200
build_flags =
    -DLV_CONF_INCLUDE_SIMPLE
    -DDISABLE_ALL_LIBRARY_WARNINGS
    ; Enable UARDUINO_USB_CDC_ON_BOOT will start printing and wait for terminal access during startup
    ; -DARDUINO_USB_CDC_ON_BOOT=1
    ; Enable UARDUINO_USB_CDC_ON_BOOT will turn off printing and will not block when using the battery
    -UARDUINO_USB_CDC_ON_BOOT

; If you are not using TFT_eSPI, and GFX Library for Arduino can be ignored to reduce compilation time.
lib_ignore = 
    TFT_eSPI
    GFX Library for Arduino
lib_deps = 
    https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB
    FS
    SPIFFS

```

[image]

lewisxhe, 2024-01-25:
Now LilyGo-T-RGB has been registered in platformio, you can also use pio management to do it
```
[env:T-RGB]
platform = espressif32@6.3.0
board = esp32s3box
framework = arduino
board_build.partitions = default_16MB.csv
upload_speed =  921600
monitor_speed = 115200
build_flags =
    -DLV_CONF_INCLUDE_SIMPLE
    -DDISABLE_ALL_LIBRARY_WARNINGS
    ; Enable UARDUINO_USB_CDC_ON_BOOT will start printing and wait for terminal access during startup
    ; -DARDUINO_USB_CDC_ON_BOOT=1
    ; Enable UARDUINO_USB_CDC_ON_BOOT will turn off printing and will not block when using the battery
    -UARDUINO_USB_CDC_ON_BOOT

; If you are not using TFT_eSPI, and GFX Library for Arduino can be ignored to reduce compilation time.
lib_ignore = 
    TFT_eSPI
    GFX Library for Arduino
lib_deps = 
    lewisxhe/LilyGo-T-RGB @ ^1.0.0
    FS
    SPIFFS

```

ThomasSlowhand, 2024-01-25:
Hello Lewis,

 

your new library is functional, after checking your files I have inserted the needed lines to inherit the class and to invoke it into my code, both round T-RGB 2.1 and 2.8 have touch functionality now – many thanks. I owe you a beer.

 

Mit freundlichen Grüßen,

best regards

 

Thomas Krim

lewisxhe, 2024-01-26:
cheers🍻

closed 2024-01-26 by lewisxhe

lewisxhe, 2024-01-26:
@ThomasSlowhand Please note that due to changes in platformio, LilyGo-T-RGB was registered under my name and has now been changed to xinyuan-lilygo/LilyGo-T-RGB.

Please remember to change the dependency to
```
lib_deps = 
    xinyuan-lilygo/LilyGo-T-RGB @ ^1.0.0
```

ThomasSlowhand, 2024-01-26:
Hello Lewis,

this seems to cause dozens of compiler warnings:

[image]

ThomasSlowhand, 2024-01-26:
Many thanks, Lewis, but do you have any clue, while serial debug is not working, even if 

-DBOARD_HAS_PSRAM
    -DLV_LVGL_H_INCLUDE_SIMPLE
    -DARDUINO_USB_MODE=1
    ; Enable UARDUINO_USB_CDC_ON_BOOT will start printing and wait for terminal access during startup
     -DARDUINO_USB_CDC_ON_BOOT=1
    ; Enable UARDUINO_USB_CDC_ON_BOOT will turn off printing and will not block when using the battery
    ;-UARDUINO_USB_CDC_ON_BOOT

is set correctly?

lewisxhe, 2024-01-27:
You can add while(!Serial); after Serial.begin(115200);
Wait for the Serial monitor to connect before starting it and see if you can print messages.
```
void setup(){
Serial.begin(115200);
while(!Serial); 

while(1){
Serial.println(millis());
delay(1000);
}
}
```

ThomasSlowhand, 2024-01-27:
Hello Lewis,

 

many thanks, I will check that and report to you. But I also have encountered a discrepancy in the library dependencies, all needed libraries are installed twice, first in the libdeps folder and a second occurrence in the Lilygo-T-RGB subfolder:

ThomasSlowhand, 2024-01-27:
I just have checked the delay to open the serial port, it do not work.

 

It seems to be, that the code is remaining in the endless loop 

 

while(!Serial); 

 

and the serial port is not opened.

 

What I have encountered:

 

If I press the reset button and open the serial port in the serial monitor by clicking “Start monitoring” shortly after release of the reset button, the serial output comes. But it does not come anymore by resetting the dive again by rest button, and it there is no error message in the serial monitor, that the serial port has been closed due to reset of the device.

lewisxhe, 2024-01-27:
I noticed ARDUINO_USB_MODE=1 , please set ARDUINO_USB_MODE to 0 , or remove it, if set to 1 then serial messages will be output from GPIO43,44 by default

ThomasSlowhand, 2024-01-27:
Setting ARDUINO_USB_MODE=0 causes warnings for each complied fiel:

<command-line>: warning: "ARDUINO_USB_MODE" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pio\build\T-RGB\lib913\lvgl\draw\sdl\lv_draw_sdl_stack_blur.c.o
Compiling .pio\build\T-RGB\lib913\lvgl\draw\sdl\lv_draw_sdl_texture_cache.c.o

Removing ARDUINO_USB_MODE=1 resolves these warnings.

But I think, it is some sort of timing problem:

If I press "RESET" on the device only shortly, the serial monitor does not recognize the disconnection of the device and stays connected. No serial output.

But if I press RESET longer than 2-3 seconds, the serial monitor recognizes the disconnection of the device and reconnects, and serial output is running.

lewisxhe, 2024-01-27:
T-RGB uses USBC, that is, USB-CDC simulates serial and then outputs messages. When RESET is pressed, the device will not be able to run. At this time, the USB device will be invalid, so the serial monitor cannot recognize it.
But after releasing RESET, the device will restart and the USB serial device should be recognized normally.
The effect of short press and long press should be the same. If short press does not work, I will suspect whether there is something wrong with the button.

ThomasSlowhand, 2024-01-27:
No, it only works when the RESET button is pressed long. No hardware failure, I've tried this with two different devices (2.1 full circle and 2.8"), same behavior on both devices. Cross-checked with another project and ESP32 with IL9341 display, this acts normal with the serial monitor.

ThomasSlowhand, 2024-01-27:
A short press on reset does not make the USB device invalid for the serial monitor. And, as already mentioned, I think the serial monitor output is much slower than normal.

lewisxhe, 2024-01-27:
I'm not sure if my understanding is correct, can you record a video for me to check if it is correct?

ThomasSlowhand, 2024-01-28:
When I press "Reset" only shortly, the serial is not disconnected from the device. Only if I press "Reset" longer than 3-5 seconds, than the serial monitor detects a connection loss.

lewisxhe, 2024-01-29:
I uploaded a video so you can see if your T-RGB behavior matches mine
https://youtu.be/8d4hLItuTR4

ThomasSlowhand, 2024-01-29:
Hello Lewis,

many thanks for your engagement, it is truly appreciated. Despite your video more looks like the necessary driver installation for a SIM-7000, my T-RGB act like described: A short press on "RESET" does not make the PIO serial monitor disconnect from the COM-port, it stays connected. Only by a long press (4-5 sec) on "RESET" the serial monitor detects a communication loss to the com-port and reconnects, after reconnection debug information by "serial.print" is displayed. When stopping serial monitor output  by clicking "stop Monitoring" the screen of the device is shut off and gets dark.

lewisxhe, 2024-01-29:
Sorry, the link is wrong, this is
https://www.youtube.com/watch?v=srqI424p5Yk

lewisxhe, 2024-01-29:
I noticed you said platformio monitor's window, this is inaccurate, you should be monitoring the computer device manager

lewisxhe, 2024-01-29:
This is the Serial monitoring software used in my video
[Sscom_V5.13.1.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/files/14081569/Sscom_V5.13.1.zip)

lewisxhe, 2024-01-29:
I rarely use platformio, but I just tested it and I have the same problem as you. This is a problem with the platformio monitor. After you press Rst, make sure you click on the Serial monitor and press Enter on the keyboard. platformio It can detect Serial disconnection immediately. If you don't press Enter, it will not be detected.

ThomasSlowhand, 2024-01-29:
> I noticed you said platformio monitor's window, this is inaccurate, you should be monitoring the computer device manager

My device manager acts the same as yours. Immediate disconnection detected.

ThomasSlowhand, 2024-01-29:
> This is the Serial monitoring software used in my video [Sscom_V5.13.1.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/files/14081569/Sscom_V5.13.1.zip)

I use this and it works fine:

[image]

ThomasSlowhand, 2024-01-29:
> I rarely use platformio Serial monitor, but I just tested it and I have the same problem as you. This is a problem with the platformio monitor. After you press Rst, make sure you click on the Serial monitor and press Enter on the keyboard. platformio It can detect Serial disconnection immediately. If you don't press Enter, it will not be detected.

What are you using? Arduino IDE? Many thanks for the hint, I will check it.

lewisxhe, 2024-01-29:
So that's normal. 
I use platformio as the compiler, but I use SSCON to monitor the serial data, because it can freely set the RTS DTR to reset the board without disconnecting the USB connection

ThomasSlowhand, 2024-01-29:
> So that's normal. I use platformio as the compiler, but I use SSCON to monitor the serial data, because it can freely set the RTS DTR to reset the board without disconnecting the USB connection

I cannot use that, because the UI is in Chinese...

lewisxhe, 2024-01-29:
[image]
Check here