Xinyuan-LilyGO/LilyGo-T-RGB #46: Analog in

ThomasSlowhand, 2024-03-13:
Is it possible to re-assign a pin on the T-RGB for enabling an Analog-in corresponding to PA12 on a Bluepill? 

(https://stm32-base.org/boards/STM32F103C8T6-Blue-Pill.html).

Background: I am thinking about connecting a atomic clock receiver

https://universal-solder.ca/product/60khz-wwvb-atomic-clock-receiver/

to the device for getting synchronized time w/o NTP.

lewisxhe, 2024-03-13:
[image]
If you must choose one, then IO0 may be the best choice. If it is used as an analog input, you need to remove the pull-up resistor. It depends on how you choose.

ThomasSlowhand, 2024-03-13:
Many thanks, Lewis, your swift reply and help is highly appreciated.

lewisxhe, 2024-03-18:
If it's ok, I close it

closed 2024-03-18 by lewisxhe

ThomasSlowhand, 2024-05-11:
Hello Lewis,

can you assist? I’d need a power consumption table for T-SIM 7600-H in all operation modes to check and plan a öong-term application with reduced power availability.

Mit freundlichen Grüßen,
best regards

Thomas Krim
Watch the Ernst Westphal TV Report!<https://ernst-westphal.de/video/Ernst-Westphal-web.mp4>

Ernst Westphal e. K.
Passion for watches. Since 1898.

German traditional wholesale supply for sophisticated watchmaking and chronometry
Deutscher Traditions-Fachgroßhandel für Feinuhrmacherei und Chronometrie

Hamburg/Germany
Handelsregister HRA 47518 – AG Hamburg
USt. ID-Nr.  DE175597301