Xinyuan-LilyGO/LilyGo-T-RGB #47: LilyGo_RGBPanel.h compile error

savall21, 2024-03-13:
When I try to compile any of the examples using LilyGo_RGBPanel.h, I get compile errors stemming from line #24 #include <ExtensionIOXL9555.hpp>.  Where is "ExtensionIOXL9555.hpp"??

lewisxhe, 2024-03-13:
Do you use platformio or arduino?
Using Arduino IDE or platformio will automatically download dependencies, ExtensionIOXL9555 is located
[ExtensionIOXL9555.hpp](https://github.com/lewisxhe/SensorLib/blob/master/src/ExtensionIOXL9555.hpp)

I ran a compile check and found no problems
https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/actions/runs/8261965632

https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/actions/runs/8261963905

savall21, 2024-03-13:
I'm using Arduino IDE. My errors:
In file included from /Users/chriscarpenter/Downloads/lilygo/LilyGo-T-RGB-main/examples/lv_factory/lv_factory.ino:10:
/Users/chriscarpenter/Documents/Arduino/libraries/LilyGo-T-RGB/src/LilyGo_RGBPanel.h:103:24: error: 'ExtensionGPIO' in 'class ExtensionIOXL9555' does not name a type
     ExtensionIOXL9555::ExtensionGPIO cs = ExtensionIOXL9555::IO3;
                        ^~~~~~~~~~~~~
/Users/chriscarpenter/Documents/Arduino/libraries/LilyGo-T-RGB/src/LilyGo_RGBPanel.h:104:24: error: 'ExtensionGPIO' in 'class ExtensionIOXL9555' does not name a type
     ExtensionIOXL9555::ExtensionGPIO mosi = ExtensionIOXL9555::IO4;
                        ^~~~~~~~~~~~~
/Users/chriscarpenter/Documents/Arduino/libraries/LilyGo-T-RGB/src/LilyGo_RGBPanel.h:105:24: error: 'ExtensionGPIO' in 'class ExtensionIOXL9555' does not name a type
     ExtensionIOXL9555::ExtensionGPIO sclk = ExtensionIOXL9555::IO5;
                        ^~~~~~~~~~~~~
/Users/chriscarpenter/Documents/Arduino/libraries/LilyGo-T-RGB/src/LilyGo_RGBPanel.h:106:24: error: 'ExtensionGPIO' in 'class ExtensionIOXL9555' does not name a type
     ExtensionIOXL9555::ExtensionGPIO reset = ExtensionIOXL9555::IO6;
                        ^~~~~~~~~~~~~
/Users/chriscarpenter/Documents/Arduino/libraries/LilyGo-T-RGB/src/LilyGo_RGBPanel.h:107:24: error: 'ExtensionGPIO' in 'class ExtensionIOXL9555' does not name a type
     ExtensionIOXL9555::ExtensionGPIO power_enable = ExtensionIOXL9555::IO2;
                        ^~~~~~~~~~~~~
/Users/chriscarpenter/Documents/Arduino/libraries/LilyGo-T-RGB/src/LilyGo_RGBPanel.h:108:24: error: 'ExtensionGPIO' in 'class ExtensionIOXL9555' does not name a type
     ExtensionIOXL9555::ExtensionGPIO sdmmc_cs = ExtensionIOXL9555::IO7;
                        ^~~~~~~~~~~~~
/Users/chriscarpenter/Documents/Arduino/libraries/LilyGo-T-RGB/src/LilyGo_RGBPanel.h:109:24: error: 'ExtensionGPIO' in 'class ExtensionIOXL9555' does not name a type
     ExtensionIOXL9555::ExtensionGPIO tp_reset = ExtensionIOXL9555::IO1;
                        ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

lewisxhe, 2024-03-13:
Which version of esp32 core are you using? 3.0?

lewisxhe, 2024-03-13:
[image]
Can you compare to see if the installation is complete?

savall21, 2024-03-13:
Thats the thing.  When I'm in ArduinoIDO and I search the Library for LilyGo-T-RGB, I don't get anything.  I imported the zip file from github

lewisxhe, 2024-03-13:
To import files, you must manually download dependent libraries instead of simply importing LilyGo-T-RGB.

https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB#5%EF%B8%8F%E2%83%A3-arduino-ide-manual-installation

lewisxhe, 2024-03-13:
You can't find it? ?

[image]

savall21, 2024-03-13:
SensorLib in the library only gives me an option up to version 0.1.4

lewisxhe, 2024-03-13:
[image]

> SensorLib in the library only gives me an option up to version 0.1.4

savall21, 2024-03-13:
[image] No, I don't see it

lewisxhe, 2024-03-13:
oh ! I don’t know what the situation is there, you can run the command line according to action CI
https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/blob/f9249cc55c71f011fab2e5a20354702727f67546/.github/workflows/arduino_ci.yml#L53

lewisxhe, 2024-03-13:
I suspect that your ArduinoIDE failed to pull the library index library, resulting in the inability to index the latest library.

savall21, 2024-03-13:
Wow, ok let me try that

savall21, 2024-03-13:
Installed the latest version of Arduino IDE and reverted to lvgl 8.3.1 and it now compiles. Thanks!

closed 2024-03-13 by savall21