Xinyuan-LilyGO/LilyGo-T-RGB #48: Wake on Touch - getting the display to work again

DrJaymz, 2024-03-15:
I need a bit of help here - I wanted to sleep and then wake from a touch event.  I wake up the screen comes back on but is all sorts of messed up.  When waking from deep sleep I was under the impression that the board effectively resets and so the screen should be initialised as on power on.

_Originally posted by @DrJaymz in https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/issues/23#issuecomment-1999866497_

lewisxhe, 2024-03-17:
Which model of touch chip do you use?

DrJaymz, 2024-03-17:
> Which model of touch chip do you use?

Its the 2.8 inch model so thats a gt911 or something (dont have it in front of me)

lewisxhe, 2024-03-18:
Examples have been added to test GT911 and CST820 versions

closed 2024-05-02 by github-actions[bot]