Xinyuan-LilyGO/LilyGo-T-RGB #49: No touch

Suneboe, 2024-03-18:
I've dowloaded the repository and opened in PlatformIO. I can build and upload all examples but the touch interface does not work for any of them. I have the 2.1" half circle with FT3267. It did work for the example already on the device from factory. What should I be looking for?

lewisxhe, 2024-03-19:
Oh. I just tested the CST820 and GT911 versions yesterday, but I didn’t test the FT3267. I don’t have this screen now. Please give me a day to investigate. The screen should be in my hands tomorrow.

lewisxhe, 2024-03-20:
The fix has been submitted, please try it

Suneboe, 2024-03-20:
Tested and it works 👍 Thank you
FYI - there's a typo in the platformio.ini. In the list of examples the "AdjustBacklight" is spelled "AdjustBlacklight" which makes it fail to find it when building

lewisxhe, 2024-03-21:
Thanks for reminding

Suneboe, 2024-03-21:
Thanks for the fast response. Commenting to close issue

closed 2024-03-21 by Suneboe