Xinyuan-LilyGO/LilyGo-T-RGB #52: UART Pin?

crackerjacques, 2024-04-11:
I was recently building a UART monitor & quick command button to connect to a microcontroller.
I can already send and receive USB-Serial, but I don't know how to use the 4-pin connector on top of the TF Card slot.
For example

`SoftwareSerial SoftSerial(GPIO_NUM_48, GPIO_NUM_8);`

If I set it like this, I get no screen, and the same thing happens if I remove GPIO_NUM_ and set the hardware pin number.

How exactly do I use this pin as a UART pin?
I would appreciate it if you could enlighten me.
[image]

lewisxhe, 2024-04-12:
There are no extra IOs used as serial ports, 48 ​​and 8 can only be used as I2C. This picture is wrong, I have reported it to LilyGo

crackerjacques, 2024-04-13:
> There are no extra IOs used as serial ports, 48 ​​and 8 can only be used as I2C. This picture is wrong, I have reported it to LilyGo

Thanks.
However, since the ESP32s3 has bluetooth and Wi-Fi, I'm going to plan B based on that.

closed 2024-05-27 by github-actions[bot]