Xinyuan-LilyGO/LilyGo-T-RGB #55: External I2C

tBubus, 2024-05-30:
Hej,

can you provide some examples how to use the external I2C?

lewisxhe, 2024-06-12:
The external I2C has been initialized in the library. Just connect the sensor to the terminal and then directly call the initialization of your sensor library to use it.

Mike-Kuznetsov, 2024-07-16:
I've made it work with my i2c BMP280 sensor.
It works even if you don't want to use Lilygo-T-RGB library.
It works even if you don't have grove connectors.

Steps:
1. Google grove pinout.
2. Solder wires (SDA to SDA, SCL to SCL, 3,3v to 3.3v, GND to GND)
3. Get an example code from your sensor's library (for example Adafruit_BMP280 lib)
4. Add "Wire.begin(8, 48);" to setup function before the code which starts sensor activity. It changes standard SDA and SCL pins to 8 and 48
5. It works

closed 2024-08-30 by github-actions[bot]