Xinyuan-LilyGO/LilyGo-T-RGB #58: Arduino IDE Sketch size too large

jsanctum, 2024-06-21:
Have tried several examples with the 2.8" TTGO T-RGB display using the settings recommended in the README. I always get a similar warning message to:

```
Sketch uses 511393 bytes (16%) of program storage space. Maximum is 3145728 bytes.
Global variables use 22340 bytes (6%) of dynamic memory, leaving 305340 bytes for local variables. Maximum is 327680 bytes.
```

Bytes for each example are different but all exceed 3145728.  

Final output:
```
esptool.py v4.5.1
Serial port /dev/cu.usbmodem2101
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 74:4d:bd:87:83:d8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600

A fatal error occurred: No serial data received.
Failed uploading: uploading error: exit status 2
```

jsanctum, 2024-07-12:
Correction, the file size is within limits. To address the `No serial data received` error:
1. Unplug USB-C cable
2. Press & hold boot button (labeled BOT)
3. Reinsert USB-C cable
4. Run upload from Arduino IDE

Output terminal now reports expected:
```
Wrote 2910592 bytes (1152447 compressed) at 0x00010000 in 23.3 seconds (effective 999.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

However, I still always get a blank screen, even after hitting the reset (labelled RST) button or powering off-on.

lewisxhe, 2024-07-12:
Write a firmware to test whether the hardware is intact
https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/tree/main/firmware

jsanctum, 2024-07-14:
Thx @lewisxhe. Reseting the firmware didn't work for my 2.8" T-RGB Round device. The backlight comes on, but even the firmware update produces a black screen.

I confirmed the above esptool firmware update process works on a different device. So I suspect it's a hardware issue at this point.

closed 2024-08-28 by github-actions[bot]