Xinyuan-LilyGO/LilyGo-T-RGB #69: LilyGo T-RGB

Lio74150, 2024-10-15:
Hi to everyone.Bought a T-RGB and have an issue.Dowload the LV image example with Arduino IDE.Pictures were badly displayed so I burned the Lv firmware.bin with succes but pictures are still grey with lines.Followed every line of the tutorial and every power sourcr.Any one have a clue?Many thanks !!!!!

[image]
[image]
[image]
[image]

lewisxhe, 2024-10-16:
This may be a damaged screen, please contact the dealer for a replacement

Lio74150, 2024-10-16:
Hi lewisxhe thanks for answer.Well the screen was fine before uploading lvimages.I bought another one  , upload lvimages and.... same thing.I upload lvfactory.bin and same problems.So I'am unlucky with 2 faulty screens , are there is a problem with driving the screen.On the second one in the "brigtness" screen there is ghosty icons of CPU , RAM and SD ( the last menu).Best regards :)

lewisxhe, 2024-10-16:
Do you use platformio or arduino IDE?

Lio74150, 2024-10-16:
I'm using Arduino IDE , but the lvfirmware.bin is upload with  "flash_download_tool_3.9.5_0"

lewisxhe, 2024-10-17:
Which version of esp32 core are you using?  Currently using 3.0 and below, everything is normal, and there is no update test above 3.0
[image]

lewisxhe, 2024-10-18:
No ! I mean downgrade to version 2.0.17 first and then test it .

Lio74150, 2024-10-18:
Fresh Windows10 , fresh Arduino IDE , ESP32 2.0.17 , LVGL 8.3.11, your LilyGo library , Lvimages.ino ....... and still no readable picture.Could it be a screen timing parameter, it's the 2.8 inches?

lewisxhe, 2024-10-21:
No readable images? Have you put the images in the [data](https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/tree/main/examples/lv_images/data) directory into the SD card?

Lio74150, 2024-10-21:
Readable is not the right word viewable is better ?Yes , it's picture 3 and 4 at the beginning of this post for exemple

lewisxhe, 2024-10-21:
Can the SD card be mounted normally?

Lio74150, 2024-10-21:
The SD card is a 8G and is ok.I tried "LV hello word" , picture is also bad so no SD issues

lewisxhe, 2024-10-21:
So now your problem is that the display is bad, right? Just like the picture you provided above, right?

Lio74150, 2024-10-21:
Saddly I think it is

Lio74150, 2024-10-21:
But 2 faulty is really a lack of lust

lewisxhe, 2024-10-21:
A patch for versions above v3.0.0 has just been submitted and the test is normal.

[image]
[image]

lewisxhe, 2024-10-21:
I personally think there may still be something wrong with your screen.

Lio74150, 2024-10-22:
Well is your test screen a 2.8 and can you upload your lvimages.bin?Last chance before saying screens are faulty.Many thanks

lewisxhe, 2024-10-22:
[lv_images.ino.merged.zip](https://github.com/user-attachments/files/17473668/lv_images.ino.merged.zip)
[image]
Then give it a try

Lio74150, 2024-10-22:
[image]
[image]
[image]
[image]

Saddly my two screens seem to be faulty.Many thanks for all the energy and time you take for me.Best regards

lewisxhe, 2024-10-23:
Is the one in the picture the same as the one I'm using? If so then it's probably broken.

Lio74150, 2024-10-23:
the .bin is yours and it's both 2.8 "

lewisxhe, 2024-10-23:
I mean the pictures in SD!

Lio74150, 2024-10-23:
it's yours as well

lewisxhe, 2024-10-23:
Okay. I have no other ideas.

Lio74150, 2024-10-23:
Thanks for all the time you spent on this issue.Best regards

Lio74150, 2024-10-23:
issue closed