Xinyuan-LilyGO/LilyGo-T-RGB #70: Problems with Programming / Examples does not work correctly

kaiuweb-coder, 2024-10-23:
Hi, my Name is Kai and I had an Issue with the 2.8 T-RGB.
I used Platform and I can build the Examples and Update the ESP works. I get also a Success. But the Display has already Stripes after the Update. No Clear Display. So that I Think that is broken. But I think also there is a Problem in the Code. I used Platform and the latest Download from the Github Repository.

Excuse the bad English but im not an native speaker.

I hope someone can help me. 
[image]
[image]

lewisxhe, 2024-10-24:
Similar to this, you can try the steps in this issue to test and see if it works properly? https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/issues/69

Lio74150, 2024-10-26:
Hi Kai , please google TL021WVC01 (2.1") , TL028WVC01(2.8") and "Timing".Maybe you'll get a clue

kaiuweb-coder, 2024-10-29:
Hi I tried the 2 Comments. Without a correction. I Think the display is broken. I send it back and hope that the next one works.

Lio74150, 2024-10-29:
Well I get no refund for the first one ( Paypal) and I guess it will be the same for the second.So as the first one is saddly mine I bought another ESP32S3 based board from well known concurrent and it s working with that LCD.So LCD is not faulty.But timing beetween 2.1 and 2.8 are clearly different.Spent to much time on it.Clearly disapointed by Lilygo this time

lewisxhe, 2024-10-30:
If it doesn't work, please contact the dealer for after-sales service.

annodomini2, 2024-11-02:
Also experiencing this, if I load the pre-built binary it works fine. So the HW is not faulty.

If I use a locally compiled binary (Arduino), I observe this.

Arduino 2.3.3
Expressif 2.0.6 Board configuration
HW settings as specified in README.md

main known issue:

Have to use TFT_eSPI 2.5.31, 2.5.22 is no longer available.

All other settings per specified except SensorLib 0.2.2, tried 0.1.17, don't think this is the issue.

Is it the TFT_eSPI library above?

Lio74150, 2024-11-02:
I get a full refund for the second one.I bought Adafruit Qualia ESP32-S3 , with the Adafruit SD card reader.You can set the screen timings , and all runs smooth.I use Arduino IDE

annodomini2, 2024-11-02:
Could you provide more information on "Set the screen timings", I have only received the screen today and I am attempting to just execute the examples.

Lio74150, 2024-11-02:
Sorry you can set timings with Qualia.Never managed with LilyGo

annodomini2, 2024-11-09:
Update: After switching to PlatformIO and VSCode, I have found only 1 example that works:

Arduino_GFX_HelloWorld

All the others result in lines, I assume this is related to the pin mapping as this appears to be different for this Sketch vs the others.

Arduino_GFX_SpriteGif does not compile, giving me font errors:
[image]

My main issue now is that the pinout for the Display ribbon is not available.

The Display is wired as per the images in the site:

[image]

lewisxhe, 2024-11-11:
@annodomini2 
I added a fixed screen initialization method. Under normal circumstances, as long as the correct touch model is identified, the screen type can be determined. However, other situations may occur. Now you can use the following four methods to initialize the screen. Please try it.
```
// Automatically determine the touch model to determine the initialization screen type. If touch is not available, it may fail.
bool rslt = panel.begin();

// Specify 2.1-inch semicircular screen
// https://www.lilygo.cc/products/t-rgb?variant=42407295877301
// bool rslt = panel.begin(LILYGO_T_RGB_2_1_INCHES_HALF_CIRCLE);

// Specified as a 2.1-inch full-circle screen
// https://www.lilygo.cc/products/t-rgb
// bool rslt = panel.begin(LILYGO_T_RGB_2_1_INCHES_FULL_CIRCLE); // Specified as a 2.8-inch full-circle screen // https://www.lilygo.cc/products/t-rgb?variant=42880799441077 // bool rslt = panel.begin(LILYGO_T_RGB_2_8_INCHES); if (!rslt) { while (1 ) { Serial.println("Error, failed to initialize T-RGB"); delay(1000); } } ```

lewisxhe, 2024-11-11:
> Update: After switching to PlatformIO and VSCode, I have found only 1 example that works:
> 
> Arduino_GFX_HelloWorld
> 
> All the others result in lines, I assume this is related to the pin mapping as this appears to be different for this Sketch vs the others.
> 
> Arduino_GFX_SpriteGif does not compile, giving me font errors: ![Arduino_GFX_Sprite_gif_build_error](https://private-user-images.githubusercontent.com/2330931/384605065-a0d9c08e-4558-4ac8-a629-4dd1b46db5ca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEyOTI1NTksIm5iZiI6MTczMTI5MjI1OSwicGF0aCI6Ii8yMzMwOTMxLzM4NDYwNTA2NS1hMGQ5YzA4ZS00NTU4LTRhYzgtYTYyOS00ZGQxYjQ2ZGI1Y2EucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTFUMDIzMDU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDgwOWM1NTAwYmZkNzI4NzkzYTRhYTFkYzg4ZWY0ZjRiNDc2MmMzM2Q1ZDAyNTQ3YzRhZjE1M2EzMzlhZmI2MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.b164wBvwAGrEQ5WWCSosSAvEpiR_VIBWChAS9pToO5U)
> 
> My main issue now is that the pinout for the Display ribbon is not available.
> 
> The Display is wired as per the images in the site:
> 
> ![T_RGB1_600x600](https://private-user-images.githubusercontent.com/2330931/384605183-a10108ad-4ea4-44e7-ad41-0cdcafab9db6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEyOTI1NTksIm5iZiI6MTczMTI5MjI1OSwicGF0aCI6Ii8yMzMwOTMxLzM4NDYwNTE4My1hMTAxMDhhZC00ZWE0LTQ0ZTctYWQ0MS0wY2RjYWZhYjlkYjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTFUMDIzMDU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjM0NDA4MmYxNzhiNjZhNzU3ZDM1ZDU1MGMyNzQ5ZjJjMGYxMmM5ZWU1OWEyMDA4NWY0YWMxNTY2MGY4NmRhZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.vBez4H3SiQ46JSflu1UUoxEnJTGoJ2JqVvcFAbm7Rlo)

This compilation error has been fixed in https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/commit/be6ebdb2a7e21665cd47dd84aba7459269efc57d

annodomini2, 2024-11-12:
Confirm Build issue resolved.

Unfortunately the linked images are not visble to me.

lewisxhe, 2024-11-13:
What image ?

annodomini2, 2024-11-14:
Apologies, my images are not appearing in the reply

lewisxhe, 2024-11-15:
Is the problem solved ?

annodomini2, 2024-11-16:
Uncertain, Backlight comes on, but no image.

Attempting to determine if there is something wrong with my setup.

lewisxhe, 2024-11-18:
@annodomini2 Try to test if the firmware works, if it doesn't work then it's broken

https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/tree/main/firmware

annodomini2, 2024-11-23:
I can see it restarting from the serial port

lewisxhe, 2024-11-25:
Arduino_GFX_SpriteGif requires the gif to be placed in the SD card. Have you done this? What about other examples?

annodomini2, 2024-11-30:
I can see that in the code, there's zero mention of it in the README.md