Xinyuan-LilyGO/LilyGo-T-RGB #72: i2c connector (molex?)

borkog, 2024-11-13:
Anyone knows what is I2C connector type?

It looks to me something like this
https://www.tme.eu/hr/en/details/mx-35507-0400/raster-signal-connectors-2-00mm/molex/35507-0400/

Can anyone confirm?

lewisxhe, 2024-11-14:
It looks right from the picture. Here is the link from Amazon for comparison

https://www.amazon.com/Seeed-Studio-Grove-Female-Conversion/dp/B074MDM36N/ref=sr_1_10?crid=2FSCJ7M87F286&dib=eyJ2IjoiMSJ9.LubXsK5rhe30w5CC5ucrZi4obkOcToWFVkxyC06Lw1eWRN8FxDllwyi-_MYuwZ2KukH1LFq7ujatWCfp_XG61OP-C1VujaZ5hAgeUeUNj_oGyoR2TRV9nrVU2cM-4mnSSy1ZhJC_EayEewk8l6NJTBRseP3LWBhtXaL4l2Zs41bniDvJb5CtR_fa9hFqfWDZMKTq-s_Uy4pWrkE2f1sy0i9q0tiq6Mu0FQLsCztSLKcH46wITjuiyYlGuJI1cJfffiIcY16ZFOGpTbC-Aufr8lSuZF-ieFi5fWlDQVbU3jU.NJnyxoEh3UPB5gTEffk8vnqsgzvRj7yw78cG0HeDBcE&dib_tag=se&keywords=Grove+line&qid=1731549238&sprefix=grove+line%2Caps%2C333&sr=8-10

hoopstarsgarage, 2024-11-25:
I've used the 4 pin version of these:

https://www.aliexpress.com/item/1005005562174022.html

lewisxhe, 2024-11-25:
@hoopstarsgarage  Thanks

borkog, 2024-11-25:
Thanks everyone!
https://www.tme.eu/hr/en/details/pap-04v-s/raster-signal-connectors-2-00mm/jst/
https://www.tme.eu/hr/en/details/sphd-001t-p0.5/raster-signal-connectors-2-00mm/jst/