Xinyuan-LilyGO/LilyGo-T-RGB #73: Blue artifacts when displaying images

Mld30240, 2024-11-14:
Hi,
Sory for my poor english, as I'm french :)
I tried T-RGB Display (2.8) with project from Nishad2m8 (https://github.com/nishad2m8/T-RGB-YT/tree/master/02-Moon-Phase-Clock-T-RGB)
All works great, except for the image rendering.
All images (mainly with greyish or blueish tints) presents blue artifacts. Maybe a roundig color problem ?
The problem appears whatever image source is used. From C array, bin image from lvgl converter and direct load from SD card, in jpg or png format.

I loaded the same image with the same version of LVGL (but LovyanGFX instead of TFT_eSpi) on another dev board (WT32SC01 plus with ST7796UI display controller, parallel rgb565) and there is no artifact.

I saw somewhere on a STM32 forum, a user had the same problem with the ST7701s display driver, and apparently solved this and said, I quote : "This was due to the pixel polarity - it was inverted". But no further explanations.

As I have not enough knolewdge on low-level lcd display driving, I hope someone could help

Thanks

[image]
[image]
[image]

lewisxhe, 2024-11-14:
This problem is caused by the fact that T-RGB uses the RGB666 pixel format screen (cannot be configured as RGB565), and esp32s3 only supports RGB565 and RGB888 pixel formats, so some pixels are missing.
https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/rgb_lcd.html

Mld30240, 2024-11-14:
OK I understand, but is there a solution to avoid this problem ?

lewisxhe, 2024-11-15:
There is currently no solution to this problem