Xinyuan-LilyGO/LilyGo-T-RGB #74: Touch on 2.8inch

borkog, 2024-12-19:
Hi,
So I couldn't properly compile examples for 2.8inch T-RGB ... Touch does not get initialised properly because 

(Other than using `bool rslt = panel.begin(LILYGO_T_RGB_2_8_INCHES);`)

LilyGo_RGBPanel.cpp is attempting to use TouchDrvCSTXXX() and when this fails, it just exits ... if you comment out everything before `_touchDrv = new TouchDrvGT911();` it works just fine

so... please see what you can do

lewisxhe, 2024-12-19:
You purchased without touch?

borkog, 2024-12-19:
No, it worked out of the box with factory sketch. When I compiled "factory" with my wifi ssid and pass touch didnt work. 
I explained what I think the problem was

lewisxhe, 2024-12-20:
Are you using Arduino IDE or platformio? I tested it and it's normal

borkog, 2024-12-26:
Arduino IDE, don't know about esp32 core (how do I find out?)

lewisxhe, 2024-12-27:
[image]
You can see the version here. And after the compilation is completed, you will also have the version numbers of all dependent libraries displayed in the compilation log window

borkog, 2024-12-27:
3.0.7 (I don't see 3.1.0 on the list)

lewisxhe, 2024-12-30:
Open all logs, and then there will be touch detection logs printed at startup. Please check whether the touch has been detected at startup.
```
Arduino tools -> Core Debug Level:  -> Verbose
```

lewisxhe, 2024-12-30:
The latest version of SensorLib will be pushed to the Arduino library manager tomorrow, v0.2.5 will solve this problem