Xinyuan-LilyGO/LilyGo-T-RGB #81: Windows Driver Compatibility

ejg3855, 2025-03-03:
I have no idea what I am doing wrong,

Plugged in out of box, everything worked well.

Flashed an example from Arduino IDE, Hello World.

Since then Windows does not recognize the device. 

I have the driver from Lilygo.cc installed, I've tried with no drivers, etc. 

[image]

I take the device out of boot mode and it just connects/disconnects over and over in windows.

ejg3855, 2025-03-03:
I've tried another pc same issues

I've also tried to flash the single firmware file and it seems to work.

lewisxhe, 2025-03-04:
No driver is required. Follow the steps below to enter download mode and the port will be fixed.

[image]

ejg3855, 2025-03-04:
Ok I had to uninstall the 9102 driver and it seems to have worked. Thanks!