Xinyuan-LilyGO/LilyGo-T-RGB #82: Screen shifts to right on heavier load

borkog, 2025-03-09:
If I have any heavier load, like downloading, complete screen shifts to right by a little ... then a bit more ... 

restart helps 

and yes ... mine flickers from time time too

lewisxhe, 2025-03-10:
What heavy load? What do you mean?
Another external power load?

borkog, 2025-03-10:
Sorry … downloading (asynctcp)

lewisxhe, 2025-03-12:
asynctcp ? Do you mean software ? asynctcp causes screen problems ?

borkog, 2025-03-12:
I understand ESP32S3 is relatively weak .. but I set asynctcp to run on the core0 (a-synchronically) ... I'm even ok for LVGL to stop responding for a second ... I even tried to get the tick running of the hardware timer... it still drifts ... is it setup of LVGL, or ESP32?

lewisxhe, 2025-03-17:
Can you provide a smaller example for me to test?