Xinyuan-LilyGO/LilyGo-T-RGB #85: ESPHOME Config for LilyGo-T-RGB

moimart, 2025-06-20:
Hi,

I'm trying to get the LilyGo-T-RGB configure properly for ESPHOME and I'm following the schematics. So far I got this:

Only showing relevant part I have so far for the display:

```
spi:
  clk_pin: GPIO33
  mosi_pin: GPIO35
  miso_pin: GPIO34

display:
  - platform: st7701s
    dimensions:
      width: 480
      height: 480
    de_pin: GPIO45

    # Control Pins from Schematic
    hsync_pin: GPIO37
    vsync_pin: GPIO47
    pclk_pin: GPIO42

    # Data Pins from Schematic (assuming standard RGB565 mapping)
    data_pins:
      # Blue: DB0-DB4
      blue:
        - GPIO44  # DB0
        - GPIO21  # DB1
        - GPIO18  # DB2
        - GPIO17  # DB3
        - GPIO16  # DB4
      # Green: DB5-DB10
      green:
        - GPIO15  # DB5
        - GPIO14  # DB6
        - GPIO13  # DB7
        - GPIO12  # DB8
        - GPIO11   # DB9
        - GPIO10   # DB10
      # Red: DB11-DB15
      red:
        - GPIO9   # DB11
        - GPIO43   # DB12
        - GPIO7   # DB13
        - GPIO6   # DB14
        - GPIO5   # DB15
```

Anything here that might be wrong? Especially around the SPI?

lewisxhe, 2025-06-21:
You cannot use ESPHOME because you need the GPIO expansion chip to transmit the display initialization sequence. ESPHOME does not have this method.

Xoliul, 2026-01-01:
Building on Bogdan's code, this is a more extensive base, works for me:

```
external_components:
  - source: github://n-serrette/esphome_sd_card
    components: [sd_mmc_card]
    
esphome:
  name: wakeclock-lilygo-t-rgb
  friendly_name: WakeClock LilyGO T-RGB

  # ADDED: Ensure power + touch reset are applied very early
  on_boot:
    - priority: 600
      then:
        - switch.turn_on: pwr_en
        - delay: 100ms
        - lambda: |-
            // nothing; just timing separation
        - switch.turn_on: touch_reset_pulse

  # ADDED: sd_mmc_card below is provided via an external component repo.
  # Requires ESPHome >= 2025.7.0 (per that repo).
  # min_version: 2025.7.0
  # ADDED: sd_mmc_card below is provided via an external component repo.
  # Requires ESPHome >= 2025.7.0 (per that repo).
  # min_version: 2025.7.0

  platformio_options:
    board_build.flash_mode: dio
    build_flags:
      -DBOARD_HAS_PSRAM
      -mfix-esp32-psram-cache-issue
    board_build.esp-idf.memory_type: qspi_opi
    board_build.psram_type: opi
    board_build.psram_size: 8MB

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  cpu_frequency: 240MHz
  framework:
    type: esp-idf

psram:
  mode: octal
  speed: 80MHz

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

  ap:
    ssid: "WakeClock Fallback Hotspot"
    password: !secret fallback_ap_password

captive_portal:

api:
  encryption:
    key: !secret api_encryption_key
  reboot_timeout: 0s

ota:
  platform: esphome
  password: !secret ota_password

### SETUP

i2c:
  id: iic
  sda: 8
  scl: 48
  scan: true

xl9535:
  - id: xl9535_hub
    i2c_id: iic
    address: 0x20

spi:
  clk_pin:
    xl9535: xl9535_hub
    number: 5
  mosi_pin:
    xl9535: xl9535_hub
    number: 4
  interface: software

### Display

display:
  - platform: mipi_rgb
    model: T-RGB-2.1
    update_interval: never

### Touch
# NOTE:
# - T-RGB 2.1" full circle uses CST820 (supported via ESPHome's cst816 driver family).
# - If you have the 2.1" half circle, it uses FT3267 instead (different driver).
#
# Start with polling (update_interval) because some CST820 setups are flaky with interrupt_pin.
# If you later identify the INT GPIO, you can add interrupt_pin: GPIOxx.

touchscreen:
  - platform: cst816
    id: ts
    i2c_id: iic
    address: 0x15            # ADDED: CST820/CST816 family is typically 0x15
    # ADDED: TP_RES is on XL9535 (IO1) per T-RGB pinmap; give the driver a reset line
    reset_pin:
      xl9535: xl9535_hub
      number: 1
    # interrupt_pin: GPIOxx   # optional; polling is usually more reliable for CST820
    update_interval: 50ms

lvgl:
  buffer_size: 20%

### Backlight

output:
  - platform: ledc
    pin: 46
    frequency: 5000 Hz  # Increased frequency for smoother PWM
    id: pwm_output2
    min_power: 0.0  # Allow full range from 0-100%
    max_power: 1.0
    zero_means_zero: true  # Ensure 0% is truly off
  - platform: gpio
    id: tp_res_low
    pin:
      xl9535: xl9535_hub
      number: 1
      inverted: true
  # NOTE: SD_CS is only relevant if you use the SD slot in SPI mode.
  # In SDMMC mode (sd_mmc_card) it is not used, so we do NOT define it as an output.

light:
  - platform: monochromatic
    output: pwm_output2
    name: "Backlight"
    restore_mode: ALWAYS_ON

### TF (microSD) slot
# SDMMC 1-bit mode (recommended on ESP32-S3 for the T-RGB TF slot)
# NOTE: This is NOT a built-in ESPHome component.
# You must add the external_components block below.

sd_mmc_card:
  id: tf_card
  mode_1bit: true
  clk_pin: GPIO39
  cmd_pin: GPIO40
  data0_pin: GPIO38

### Battery sense (raw)

sensor:
  - platform: adc
    pin: GPIO4
    name: "Battery Sense (raw)"
    attenuation: 11db
    update_interval: 10s

### Optional debug switches (internal)
# (Handy while bringing the board up; keeps HA tidy.)

switch:
  # CHANGED: Make PWR_EN come up ON automatically (touch/display power gate)
  - platform: gpio
    id: pwr_en
    name: "T-RGB PWR_EN"
    internal: true
    restore_mode: ALWAYS_ON
    pin:
      xl9535: xl9535_hub
      number: 2

  # ADDED: SD_CS debug switch (only useful for SPI-mode SD; not used by sd_mmc_card)
  - platform: gpio
    id: sd_cs_debug
    name: "T-RGB SD_CS (debug)"
    internal: true
    pin:
      xl9535: xl9535_hub
      number: 7

  - platform: template
    id: touch_reset_pulse
    internal: true
    optimistic: true
    turn_on_action:
      - lambda: |-
          // template switch used only to run the sequence below
      - output.turn_on: tp_res_low
      - delay: 50ms
      - output.turn_off: tp_res_low

```