Xinyuan-LilyGO/LilyGo-T-RGB #87: Full circle 2.1 inch no touch

Bogdanel, 2025-07-25:
Hello, I have recently bought a 2.1 inch and then flashed it with Touchpad.ino to test the touchscreen and it seems that it doesn't work. 

https://github.com/user-attachments/assets/56c8badb-e8ba-4dbe-bf0c-f2c0c550016c

Any idea how to get this device working?

lewisxhe, 2025-07-25:
OK. If so, it may be really damaged. Please contact the dealer here, send them this link, and they will resend you the screen.

PS: I have seen the communication record between you and the dealer this afternoon. You just need to send them this issue link, and I will agree to resend you the screen for testing.

Bogdanel, 2025-07-25:
Thanks for assistance and quick replies.

closed 2025-11-01 by Bogdanel